microscope-cockpit/cockpit #586: Secondary window state not properly stored.

iandobbie, 2020-04-23:
The secondary window state should be saved and restored on the next startup.This appears to be broken for the touchscreen.

carandraug, 2020-04-24:
I took a look at this and found two issues related to this:

1. the cockpit at start of cockpit that decides whether to show or not a window, only hides the windows if they are not meant to be shown. This means that the code expects the windows to start shown. In the case of the touchscreen window, this was removed in 8d66ac9c5f83 to "fix wandering focus issue" which does add the "TODO - determine if we need to Show the touchscreen or not".

2. the code that saves the window state is a bit weird and has a few issues on its own but the main issues is that it stores the window state as shown, only if it's made being shown from the drop down window menu after having being minimized. That issue seems to be there since the beginning in fe54669890e2.

I have a fix for them both and will push them after cleanup.

carandraug, 2020-04-24:
I have fixed this with a series of commits 942f3b17cdaa..a60bf4c74b9c

I have also changed the saving of window state user config (now renamed to 'Show Window') to happen only cockpit exits since it was not working one simply closed the window on the window title bar.

closed 2020-04-24 by carandraug