microscope-cockpit/cockpit #588: Light Paths button text size wrong.

iandobbie, 2020-04-27:
The light paths button uses the OptionsButton function and sets the text to a different size, I'm not sure how as it does call wx.Button in the function

carandraug, 2020-04-27:
This issues comes from an Wenjie email where it shows that on his system the "Ligh path" font looks off. His screenshoot:

[image]

On my and Ian's system this looks right and I would guess that on Mick's system did as well.

This is caused because of the use of `cockpit.gui.device.DEFAULT_FONT` in `cockpit.gui.device.Label` which is set to wx.Font(12, wx.DEFAULT, wx.NORMAL, wx.BOLD). This can be fixed by modifying the font by getting the control font and modifying in a relative manner, like so:

```
# instead of DEFAULT_FONT
self.SetFont(self.GetFont().Bold())
# instead of SMALL_FONT
self.SetFont(self.GetFont().Smaller()) # maybe call Smaller() twice?
```

Note that in theory, and from reading the documentation, we should be able to use `self.GetFont().MakeBold()` but that does not work, the documentation comes from the C++ because in C++ `GetFont()` returns a pointer to the font but on Python it seems like we get a new instance.

carandraug, 2020-05-04:
Adding @wenjieliu11, who reported this initially, to the discussion.

wenjieliu11, 2020-05-04:
The same suggestion... "You can ask several other Windows users to see if they have this gray bar problem or not. If all not, I think it should be fine as my laptop has some strange small bugs occasionally" 
But I have to say, it is still too weird for me, the bugs of my laptop I noticed before are just relevant to the installation of some software (successful or not) and not relevant to the displaying, such as the "gray bar" issue and this text size issue.

iandobbie, 2020-05-05:
I have a windows 10 PC here that I can try it on.

carandraug, 2020-05-12:
This button does not exist anymore so the issue is "fixed". Also, the root of the issue of using hardcoded font sizes was also fixed earlier on 5319c06b650a . Since the font is now inherited from the parent I guess the issue would have been fixed if the button still existed though.

Closing as fixed.

closed 2020-05-12 by carandraug