microscope-cockpit/cockpit #589: suggested reordering of the main window buttons.

iandobbie, 2020-04-27:
Wenjie suggests reordering the buttons along the top of the mainwindow to

Abort - snap - video mode - objective - light path- single stie - multi-site

carandraug, 2020-05-04:
Adding @wenjieliu11, who reported this initially, to the discussion.

wenjieliu11, 2020-05-04:
One thing I have to clarify is that this is just a suggested possible order. You can also arrange them if there are more logical orders. The core idea of my suggestion is "these icons should be divided into different groups according to their property and the icons of the same group should be put together", i.e., classification.

iandobbie, 2020-05-05:
I agree reordering would be a good thing. The current order is pretty arbitrary. I would maybe put it as 
abort - snap - video - single site- multi site - objective - light path, 
and maybe change the name of light path to channel? This puts the camera buttons, then the experiments, then the config buttons, just still makes the channel button easy to find as it si the last, this is probably the most used config button as it allows you to switch cameras, filters, light sources and powers etc... all in one click.

carandraug, 2020-05-12:
@ChrisChong8 : @iandobbie tells me that you are working on this. When you are finished, please just mention here the name of the branch where you made your work so that we can review it.

ARCFaria, 2020-05-30:
I have a commit that solves this issue. Check my branch 589-reorder-buttons.

carandraug, 2020-05-31:
The commit looks good to me. @iandobbie can you comment on the location of the "View last file" button? I don't have a strong opinion about the order of any of the buttons.

iandobbie, 2020-06-01:
On the mac the recent changes to these buttons lokk horrid, as they are now not all the same height. Additionally what has happened to the "light path" button? I know I suggested a rename but I think it is essential functionality.

iandobbie, 2020-06-01:
[image]
The previous versions had every button with 2 lines of text so the same vertical size. I think we should also have the channels button, but I thought we were just moving the load a save channel functionality to the menu bar, not the whole channel functionality.

carandraug, 2020-06-01:
> The previous versions had every button with 2 lines of text so the same vertical size.

Do you know what recent changes caused it? It looks fine on my side. Did you also upgrade wxPython?

> I think we should also have the channels button, but I thought we were just moving the load a save channel functionality to the menu bar, not the whole channel functionality. 

I don't think we should have them back on their previous form. They are now on the menu bar, exactly the same number of mouse clicks away: 1 to open the menu, another to select the channel. This was discussed on #597, so let's continue that discussion there.

ARCFaria, 2020-06-05:
> > The previous versions had every button with 2 lines of text so the same vertical size.
> 
> Do you know what recent changes caused it? It looks fine on my side. Did you also upgrade wxPython?

 I have the same problem with both version 4.1.0 and 4.0.7, on mac.

carandraug, 2020-06-05:
@iandobbie I have opened #619 for the buttons not being expanded and as discussed on #597 the channels button is not coming back as a single button on the top row of buttons. Can you comment on the location of the "View Last file" button which has not been mentioned so far?

carandraug, 2020-06-08:
I have rebased and merged @ARCFaria proposed fix. All other issues mentioned on the discussion now have their own issue. Closing this as fixed.

closed 2020-06-08 by carandraug