microscope-cockpit/cockpit #590: Objective section is out of place in main window.

iandobbie, 2020-04-27:
The objective section lable has a different format to the other buttons on the top row of the main window and looks out of place. Why is it even on the top line?

mickp, 2020-04-27:
That's where decided to put it after we got rid of the Objectives window
because it guarantees that it is always in the same place. It could also go
at the beginning of the second line, but I think we decided that may the
line too long for Diamond where that line needs to accommodate 2 cameras,
the filter wheels, the SLM and the Linkam controls. I suggest leaving the
objectives control where it is, but making it a regular button that
displays a popup menu, and updates its own contents to display the
currently-selected objective on the 2nd line of text.

carandraug, 2020-04-27:
To expand on what was said, the location of the objective drop down menu is dependent on the number of objectives. If there is only one objective, it goes on the top row together with the other buttons. If there are more than one objective it goes on the second row with the other non-light device controls. This was done in 43fc89aa0276

iandobbie, 2020-04-28:
I think it should probably be in the top line like the light paths button, but with label text which includes the objective name. Its not a device like a camera or SLM with settings, enable/disable etc so it would look rather lost on the second row.

carandraug, 2020-05-04:
Adding @wenjieliu11, who reported this initially, to the discussion.

carandraug, 2020-08-27:
> To expand on what was said, the location of the objective drop down menu is dependent on the number of objectives. If there is only one objective, it goes on the top row together with the other [...]

Correction, it was not the number of objectives, it was based on the number of objective devices (each objective device would handle multiple objectives). Anyway, since 7e08699370b0c18 this is no longer true and the objective menu always stays on the top row.

Anyway, it looks like we're not doing anything about this, and things are as we want them to be, so closing this as invalid.

closed 2020-08-27 by carandraug