microscope-cockpit/cockpit #591: Light path button on main window does not work (or not clear how?)

carandraug, 2020-04-28:
None of the buttons on the Light Path "button" of the main window seem
to work.  All fail, each in their own manner:

The "New" button fails with:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 549, in <lambda>
    b.Bind(wx.EVT_TOGGLEBUTTON, lambda evt, b=b, c=callback: self.onButton(b, c))
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 561, in onButton
    callback()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 140, in <lambda>
    self.setPath(n)),
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 285, in setPath
    self.createNewPath()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 294, in createNewPath
    pathName = cockpit.gui.dialogs.getNumberDialog.getNumberFromUser(
AttributeError: module 'cockpit.gui.dialogs' has no attribute 'getNumberDialog'
```

The "Update" button fails with:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 549, in <lambda>
    b.Bind(wx.EVT_TOGGLEBUTTON, lambda evt, b=b, c=callback: self.onButton(b, c))
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 561, in onButton
    callback()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 140, in <lambda>
    self.setPath(n)),
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 287, in setPath
    events.publish('load exposure settings', self.paths[name])
KeyError: 'Update'
```

I'm not sure about the "Load" button since I don't have a light path
file to test, but when the open file dialog is cancelled it errors with:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 549, in <lambda>
    b.Bind(wx.EVT_TOGGLEBUTTON, lambda evt, b=b, c=callback: self.onButton(b, c))
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 561, in onButton
    callback()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 140, in <lambda>
    self.setPath(n)),
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 277, in setPath
    self.onLoadExposureSettings()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 347, in onLoadExposureSettings
    self.pathButton.setOption(self.currentPath)
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 540, in setOption
    self.mainButton.SetLabel(optionName)
TypeError: Button.SetLabel(): argument 1 has unexpected type 'NoneType'
```

The "Save" button fails with:

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 549, in <lambda>
    b.Bind(wx.EVT_TOGGLEBUTTON, lambda evt, b=b, c=callback: self.onButton(b, c))
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 561, in onButton
    callback()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 140, in <lambda>
    self.setPath(n)),
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 274, in setPath
    self.onSaveExposureSettings(self.currentPath)
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 324, in onSaveExposureSettings
    defaultFile=name+'.txt',
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

The interface is a bit weird and after clicking on it, we get
another panel of buttons which can't be dismissed.

I can't really get at fix it since I don't know what it's supposed to
be doing.  In the only issue about it (issue #510), Mick says the
same, that it's not clear what they're supposed to be doing.

@iandobbie , can you expand on the use case for the light path button
and how to make use of it?  Any sample light path file to test?

iandobbie, 2020-04-28:
I assume someone has removed the get number dialog thinking it was unused!

If you can't create a light path then the rest of the functionality is not going to work.

iandobbie, 2020-04-28:
Nope, the import just doesn't work, however fixing that still doesn't let it work. I will investigate.

carandraug, 2020-04-28:
> I assume someone has removed the get number dialog thinking it was unused!

I have checked the logs but that is not the case, there was never an import for getNumberDialog on the mainWindow file, not even d3f0ce24a41ec69af7f1 which added this feature. My guess is that it just worked because it was imported on some other module.

> If you can't create a light path then the rest of the functionality is not going to work.

Sure, but even if we fix the import, the other options still don't work. I could take a go at fixing this but I don't know how this is meant to be used.

iandobbie, 2020-04-28:
If you set some config, eg camera 1 active, light one at power x, filterwheel pos 3, you can create a new lightpath say "GFP". if you later select this from the list it should reset everything to the defined conditions. 

```
from cockpit import events
setting=dict()
events.publish('save exposure settings', setting)
setting
{'south': False, 'south-filters': 0: DAPI, 430, 'west': True, 'west-filters': 0: DAPI, 430, 'transmitted': {'isEnabled': True, 'exposureTime': 100}, '488nm power': 0.01, '488nm': {'isEnabled': True, 'exposureTime': 100}, '550nm power': 0.01, '550nm': {'isEnabled': False, 'exposureTime': 100}, 'objective': '40x'}

```
So that bit works, but the filter position can't be serialised and the restore function isnt working.

iandobbie, 2020-04-29:
I did some work on this today and din't really make any progress. The above code fragment in the python shell will return the settings. I tried to make the filterHandler just save the position but restoring this breaks things horribly as the filter handler expects a filter object and not just a position. I think we need the load routine to take the filter number and find the right object and return that.

carandraug, 2020-04-29:
> If you set some config, eg camera 1 active, light one at power x, filterwheel pos 3, you can create a new lightpath say "GFP". if you later select this from the list it should reset everything to the defined conditions.

I see. Maybe Light path is not the best name for this, it suggests that it changes the light path which is why Mick on issue #510 thought that this only flipped mirrors. Maybe something like imaging settings?

Is it expected that a user will load new and save this multiple times during a normal session? Or is it expected that there's a predefined set of settings that are used often and we want to make easy to switch between them?

iandobbie, 2020-04-29:
You are right light path probably isnt the best name, maybe channel config? Or similar. The idea is that you can set light sources, exposure times, cameras and filters and store this so you don't have to redo all these settings to switch between different configs. In my experience I usually have only set it for the current session. I have never used the save and load functionality except to prove that it worked. Obvious this is no longer true.

I have used this a lots at Diamond where flipping between transmission and fluorescence involves changing filters, light sources, exposure times and cameras. Having one button rather than about 5 makes it easier and also less likely to forgot something important.

iandobbie, 2020-04-30:
I have done some work and the filter loading ans saving vaugely works but causes segv crashes. I assume that it is the control update triggered by 

handlers/filterHander.py: 99
        self.addWatch('lastFilter', lambda f: ctrl.SetSelection(ctrl.FindString(str(f))))

I think this needs a call in main thread wrapper, but I cant work out how to do this.

iandobbie, 2020-04-30:
Camera enable state is stored and loaded again but the mainwindow control is not updated.

iandobbie, 2020-04-30:
Camera state fixed by https://github.com/MicronOxford/cockpit/commit/3370ebb4bc432eebf2009cc2b8eb3e991165f826

iandobbie, 2020-04-30:
Fixed same issue in the lightsource handler with https://github.com/MicronOxford/cockpit/commit/184aa836ad719a39a9ae77f55c14b069b5119a52

iandobbie, 2020-04-30:
The event also trigger the linkam stage which I can't easily currently check. I know this has been an issue in the past as the stage doesn't return the current condensor light state which is what we want to set.

iandobbie, 2020-04-30:
Final state stored is the objective which appears to work fine, not sure why the filterwheel code crashes the interface with a segv. This is the final issue I believe.

iandobbie, 2020-04-30:
I tried to fix the filter handler crash with a @callInMainTread directive but this made the filter changes work but clicking on say the camera control after a change of setting through the light path control lead to a segv. 

Using a wx.CallAfter wrapper made it happen less often but it still appeared to happen on occasions which implies to me that it is a threading issue. Maybe all the control updates need a wx.CallAfter wrapper?

iandobbie, 2020-05-04:
Now moved the call in mian thread to only wrap the update after move call and this fixes the crashes. However the light source on/of code doesn't work it seems to always be toggling them. I will investigate.

iandobbie, 2020-05-04:
I think I have finally sorted all this out. It now does what I would expect, "new" prompts for a name and saves the exposure settings under that name, load and save, store the data to a txt file with the channel name, or load it.  Once a channel is added it can be selected and the saved, light sources, powers and exposure time, cameras, objective and filters are selected. Other than maybe a change of name to Channels I think it is ready to be pulled. 

Maybe the button could be Channel\nconfig by default and then become Channel\nname once a new channel is created or selected, keeps it 2 characters high.

iandobbie, 2020-05-05:
I also suggest that we change the storage a retrieval slightly. If we have a menu bar we could have the main commands in this and the button just allows access to the already created channels. Then  the channels could be saved into peoples config files so they get back the same channels the next time they start up that instrument. We would need a few menu functions:

New - create a new channel with the current config and name it
delete - remove channels from list and config file. 
export - save to disk outside config file
import - open from disk outside config file

iandobbie, 2020-05-05:
I pulled my changes which actually worked in the end

carandraug, 2020-05-07:
After the changes you made I still see errors with "Load":

```
Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 545, in <lambda>
    b.Bind(wx.EVT_TOGGLEBUTTON, lambda evt, b=b, c=callback: self.onButton(b, c))
  File "/home/carandraug/src/cockpit/cockpit/gui/device.py", line 557, in onButton
    callback()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 139, in <lambda>
    self.setPath(n)),
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 275, in setPath
    self.onLoadExposureSettings()
  File "/home/carandraug/src/cockpit/cockpit/gui/mainWindow.py", line 371, in onLoadExposureSettings
    if self.lightList is not None:
AttributeError: 'MainWindow' object has no attribute 'lightList'
```

And indeed there is no `MainWindow.lightList`.  Looking in the logs, this must be broken since early 2019 when `lightList` was removed in c8b5969f3048 .

However, that piece of code is accessing the GUI elements to enable/disable light sources. I think the logic should flow the other, i.e., the code should be enabling the light which would trigger the GUI to be updated.

carandraug, 2020-05-08:
Turns out the issue was that on load, the code was changing which light sources were displayed if cockpit was a light list mode (which was a different mode when there's more than 7 light sources). Since this no longer exists, the code was just failing. With 888016c62f28 I have removed a bit more of the remaining light controls code.

Closing as fixed.

closed 2020-05-08 by carandraug