microscope-cockpit/cockpit #592: grey bar in the middle of main window

carandraug, 2020-04-28:
Reported by Wenjie Liu by email.

There is a dark grey bar across the main window, between the cameras and lights panels.

[image]

carandraug, 2020-05-02:
I have looked at all the places where the background colour is set manually and all usage of hardcoded colours but couldn't find anything that could explain this from just reading the code.

By the way, the white background on the top half of the main window was a separate issue already fixed in 26ef5e911d3a (Wejie reported by email that it's already fixed). The weird thing is that the issue was supposedly GTK3 specific and so Wenjie should not be experiencing it under Windows. It's possible that Wenjie is somehow running wxGTK instead of wxMSW.

carandraug, 2020-05-04:
@wenjieliu11 : can you take a look at this? I don't know how to replicate this issue you found.

wenjieliu11, 2020-05-04:
I just check the interface again. The gray bar still exists (following screenshot). But it may be fine to keep it the way it is as the whole background is gray now?
And sorry, what does "GTK3"/"wxDTK"/"wxMSW" mean? If you can help me to figure it out, I can check which one I am running.
Another solution may be that you can ask several other Windows users to see if they have this gray bar problem or not. If all not, I think it should be fine as my laptop has some strange small bugs occasionally...
[image]

carandraug, 2020-05-04:
> And sorry, what does "GTK3"/"wxDTK"/"wxMSW" mean? If you can help me to figure it out, I can check which one I am running.

While running cockpit, open the Python shell (right click on the main window background, and then select "PyShell > Show/Hide"). Then see what the following returns for you (this is what I get):

```
>>> import wx
>>> wx.Platform
'__WXGTK__'
```

wxWidgets is a GUI toolkit that wraps the system native GUI toolkits. The GUI toolkits on Linux like GTK and Qt can actually be installed on other operating systems like Windows. I was wondering if maybe you have GTK installed on your windows and wxWidgets is using it instead of the windows native one.

wenjieliu11, 2020-05-04:
Oh, thanks for your explanation!
I checked it, luckily and unluckily, my computer also returns '__WXMSW__'.

carandraug, 2020-05-05:
@wenjieliu11 if I was you, to find where this issue comes from, I would try to comment out chunks of code from the `MainWindow.__init__` method (on the cockpit/gui/mainWindow.py file) until you find where that grey bar is coming from.

wenjieliu11, 2020-05-06:
Hi David,
According to your suggestion, I did have some finding.
I found that in fact, it is not a "bar" between light and camera panel. The following interface is the result after commenting out the code I pointed out.  
[image]
If you drag the main windows down, you will find that the whole background is grey (following image). Therefore, I think this issue is because the light and camera panel aren't right next to each other, i.e., there is some space between these two panels when you arranged them which leads to the leakage of the "grey background" bar. 
[image]

carandraug, 2020-05-06:
> If you drag the main windows down, you will find that the whole
> background is grey (following image). Therefore, I think this issue
> is because the light and camera panel aren't right next to each [...]

It is possible.  You should bee able to confirm this by commenting out
the code that adds a space between those two panels, like so:

```
--- a/cockpit/gui/mainWindow.py
+++ b/cockpit/gui/mainWindow.py
@@ -211,7 +211,7 @@ class MainWindow(wx.Frame):
         topPanel.SetSizerAndFit(topSizer)
 
         self.Sizer.Add(topPanel, flag=wx.EXPAND)
-        self.Sizer.AddSpacer(ROW_SPACER)
+#        self.Sizer.AddSpacer(ROW_SPACER)
 
         ## Panel for holding light sources.
         self.Sizer.Add(mainPanels.LightControlsPanel(self), flag=wx.EXPAND)
```

Can you confirm that the patch above removes the dark grey bar?

However, the space between the two panels is by design.  The problem
is not that the space exists, the problem is that the background for
some reason has a different colour since all the panels should have
the same colour.  There is also the same space between the top row
buttons and the camera panels and you don't get a dark grey bar there.

wenjieliu11, 2020-05-06:
Oh, the commenting of the following code can indeed remove the space and thus the grey bar 
[image]

carandraug, 2020-05-06:
I have asked on freenode's #wxwidgets and seems like the problem is that the main window is a naked frame, .i.e., we add a sizer to it with its multiple panels. Instead, we should have a single frame, the main window panel, and set the sizer on it. This is explained on wxWidgets FAQ entry ["Why are backgrounds of windows ugly, dark grey on Windows? I want them like my dialog boxes."](https://wiki.wxwidgets.org/WxFAQ#Why_are_backgrounds_of_windows_ugly.2C_dark_grey_on_Windows.3F_I_want_them_like_my_dialog_boxes.)

And here's a redacted version of the conversation on the channel:

> DavidGH: Especially on Windows, it avoids varies issues, colour included, NOT to have a naked frame
> DavidGH: Instead give the frame a wxPanel as its only child. No need for a sizer
> DavidGH: a sole child in a frame is automatically resized to fill its client size
> DavidGH: Then do everything else inside that panel's sizer, and parented by the panel
> [...]
> carandraug: reading that faq, so default background colour of a frame on windows is different from the default background colour of a panel?
> [...]
> DavidGH: IIUC a frame doesn't _have_ a background colour on Windows, so you get to see whatever's behind the frame

carandraug, 2020-05-06:
@wenjieliu11 can you try my "wip-592-mainWindowpanel" branch and see if that fixes the issue? To do it, do the following:

```
git remote add david git@github.com:carandraug/cockpit.git
git fetch david
git checkout david/wip-592-mainWindowpanel
```

before starting cockpit. After testing, don't forget to go back to the normal branch, like so:

```
git checkout master
```

wenjieliu11, 2020-05-07:
Oh, that is interesting.
I tried these commands, but it seems that the issue is not fixed. Or maybe I didn't do it rightly because I found that the cockpit was run from the same path in my computer in these two branches. Should they be different?
The following interface is the running result after switching the branch.
[image]
BTW, to switch to your "wip-592-mainWindowpanel" branch, I need to set a SSH key first. Otherwise, it will remind 
`git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.
`
after running
`git fetch david`.

iandobbie, 2020-05-07:
Yes if you have ssh as your access mode, the URL starts with git@, then you need a valid ssh key even for public repositories. ~You can change to http access by changing the start of the URL from "git@github.com:" to "https://github.com/" I suspect that you were not actually running the new code.

wenjieliu11, 2020-05-07:
Yes, I agree with you @iandobbie 
As the command above didn't let me run the new code somehow, I downloaded your revised version manually from the "wip-592-mainWindowpanel" branch and covered the original version in my computer. And it does work!
[image]

wenjieliu11, 2020-05-07:
And it seems that you also reduced the window numbers if using your 592 branch code. There is only one main window after opening the cockpit. The other windows, mosaic, stage. etc., can be shown by right-clicking in the main window.

carandraug, 2020-05-07:
Thank @wenjieliu11 for testing. I've cleaned up the patches a bit more, squashed them, added a nicer commit message, and pushed 73d1b6c28126 to fix this issue.

I'm closing this as fixed.

> And it seems that you also reduced the window numbers if using your 592 branch code. There is only one main window after opening the cockpit. The other windows, mosaic, stage. etc., can be shown by right-clicking in the main window.

That's only because with that branch you also get the other changes pushed to master earler this week. What you see was done as part of issue #593 . The windows that you get at start of cockpit should be the same that were open when you last exited cockpit.

closed 2020-05-07 by carandraug