microscope-cockpit/cockpit #593: remove concept of primary and secondary windows

carandraug, 2020-04-29:
Cockpit has the concept of primary and secondary windows. Primary windows are never closed, the idea being that it does not make sense to close them, and the window menu they can only be moved around. Secondary windows can be closed (although there is only one instance of them and closing just hides them instead).

The problem with this is that we decide what are the primary windows not the actual user. As part of the window reduction we now changed our mind and no longer think the logging window should be a primary window (issue #575). For use in a lightsheet microscope we spoke about making the mosaic window optional (issue #547). We also spoke recently on a meeting of having a command line option that would make the new touchscreen window the main window instead of the current main window.

I propose that we just remove the whole secondary and primary window stuff. Instead, there is only the main window (which one can be an option later when the new new touchscreen window is ready). In addition, we have a window menu on the top bar menu (menu bar proposed on issue #583), common on programs with multiple windows such as gimp.

carandraug, 2020-04-30:
> I think that we should also have a default set of visible windows.

Yes. I was thinking of a config option with the current primary windows as shown by default. I'm not keen on making much changes to the touchscreen since it seems that Danny is redoing it anyway.

iandobbie, 2020-05-01:
No don't change the touchscreen at all as Danny has done massive amount of work which is still in progress and I think will on mass replace the existing touchscreen code. But we could easily change the startup code to show or hide this window on startup with no, or very little change to the touchscreen code.

carandraug, 2020-05-01:
I have a proposal implementation on my "wip-remove-secondary-windows-issue-593" branch. All windows, except the main window, can now be closed (they're actually hidden), and all have the appropriate options on the marshall windows menu.  The change is mostly about removing code.

For now, user config entries decide whether to show or hide the windows (which means it will be show or hide like at the end of the last cockpit run), but if this moves forward I have some ideas about making this more configurable.

iandobbie, 2020-05-04:
Can we add my change which unhides the logging window on output to stderr? I cant find the commit so maybe I lost it, but I can easily recreate it.

carandraug, 2020-05-04:
> Can we add my change which unhides the logging window on output to stderr? I cant find the commit so maybe I lost it, but I can easily recreate it.

That's on issue #575 but making logging window pop up when it receives new messages is a separate thing from removing primary/secondary separation and it will be made simpler after fixing this, which is why I was hoping to have this reviewed and merged first.

carandraug, 2020-05-06:
This is now done. Whether a window is show when cockpit starts is decided like so:

1. if there's a user config entry, use it (this is effectively the state from the last time cockpit ran)
2. check the new optional window property `SHOW_DEFAULT`
3. if none of the above, window is hidden

To not have any changes at the moment, I have set each window `SHOW_DEFAULT` value whether the windows was previously a primary or secondary window.

Closing as fixed.

I think in the future we could have some sort of "window.conf", similar to the "depot.conf" file, which could define this value (as well as other configuration values), and even which windows get created at all (at the moment all windows get created even if they are never used and hidden by default).  Being able to specify the class on the config file, would also work as some sort of plugin since we wouldn't have to list the windows on the code itself. I have looked a bit into this and before that is done, we need to remove the need for singleton windows.

closed 2020-05-06 by carandraug