microscope-cockpit/cockpit #594: macro stage window title does not fit correctly

carandraug, 2020-05-04:
The macro stage window title is used to name the individual macro stages, like so:

[image]

However, if the window is resized then the title no longer aligns with the individual macro stages within the frame:

[image]

Instead of abusing the window title, we should instead be labelling each macro stage. This would enable in the future to have them being resized individually. The problem with that is that a label would require space within the frame and so less space to draw the actual macro stages. But maybe the labels are not needed since they're self-explanatory (or maybe I'm just already used to it).

Removing this abuse of window title to name the panels within the frame would also remove code from the keyboard module to handle very long window titles.

iandobbie, 2020-05-04:
How about putting the positions (eg XY and Z) above the relevant sections, then we dont take up any more space but key people into what each section might represent?

carandraug, 2020-05-05:
> How about putting the positions (eg XY and Z) above the relevant sections, then we dont take up any more space but key people into what each section might represent?

Sounds good to me. @dstoychev do you want to do this since you're already making changes to the macro stage to fix issue #585 ?

carandraug, 2020-05-05:
In the mean time, unless someone has an issue with it, I will change the window title to just "Macro Stage", since it makes fixing #593 a bit easier.

carandraug, 2020-05-13:
While reviewing #587 (which mixed a proposal to fix this by moving the coordinates text to the top), why not have the the text showing the stage coordinates with plain wx text controls?

carandraug, 2020-05-13:
@dstoychev can you review my `wip-594-text-coords-in-wx` branch? It's just an rough sketch for now and only does it for xy.

dstoychev, 2020-05-14:
I can't see the branch, is it public?

carandraug, 2020-05-14:
> I can't see the branch, is it public?

Yes it is public but my repo ("fork" is not linked to avoid opening github pull requests. See https://github.com/carandraug/cockpit/tree/wip-594-text-coords-in-wx .

dstoychev, 2020-05-15:
I had a look. Here is some feedback.

commit 69400954cf55ac25ea9c5bf0b6463872f6d858c6:
- cockpit/gui/macroStage/macroStageWindow.py:
    - On line 124, I think it would be better to use the respective wxPython classes instead of implicit conversion from tuples. In this case that would be ``wx.GBPosition()`` and ``wx.GBSpan()``.
    - On lines 127 and 131, the same thing but with ``wx.Size()``.

commit cddf2059e7bc3d3c80df132490ac536bc32237d8:
- cockpit/gui/macroStage/macroStageWindow.py
    - On line 106, is it necessary to have the print() call? If yes, then perhaps the same can be added to the ``_OnUpdateStepSize()`` method.
    - The way the labels are done with StaticText now, there needs to be extra space between the position and the step. See how it looks now for me: [image] This could be done with the ``AddSpacer()`` method or even with a border on the left of the step label. Also the highlighting (in different colour) of one of the positions is now lost and should find its way back to the code. I think it may be worthwhile considering a TextCtrl control, which supports rich text and therefore you can pass a single string per axis or even for all the axes. The downside is that if you make the TextCtrl read-only, as to prevent users from accidentally deleting the contents, then there is a grey-ish background applied to the control and this doesn't look very good in my opinion. But maybe there is a workaround for this. Of course, StaticText could still be used, it's just that the colour of the particular position needs to be altered.

carandraug, 2020-05-17:
> I think it would be better to use the respective wxPython classes instead of implicit conversion from tuples. In this case that would be `wx.GBPosition()` and `wx.GBSpan()`. [...] the same thing but with `wx.Size()`.

In general I agree with the explicit being better but in this case we get there by being explicit on the argument name, i.e., `sizer.Add(panel, pos=(0, 0), span=(8, 4))` no need for `sizer.Add(panel, wx.GBPosition(0, 0), wx.GBSpan(8, 4))`. At the moment is neither.

>  is it necessary to have the print() call? [...]

No. It was only used for debugging, it will be removed in the end.

> [...] there needs to be extra space between the position and the step. [...]  the highlighting (in different colour) of one of the positions is now lost [...]

Yes. The aim was just to roughly show the idea of using wx instead of opengl for this stuff and see what you think about it before styling.

> I think it may be worthwhile considering a TextCtrl control, which supports rich text and therefore you can pass a single string per axis or even for all the axes. The downside is that if you make [...]

I think using `TextCtrl` is a very good idea since it displays nicely that this is something that may change. `StaticText` is much better for labels. However, I don't think the grey-ish background of being disabled is a bad thing, it clearly shows that it's not something the user sets by typing a number on it. Which brings another idea (for later), of a control to just type the coordinates. In the mean time, I will try a disabled `TextCtrl` and see what we get.

iandobbie, 2020-05-18:
>  Which brings another idea (for later), of a control to just type the coordinates. In the mean time, I will try a disabled `TextCtrl` and see what we get.

This functionality is in a right click menu at the moment. I think having ti front and central isn't a great idea as it could easily be very destructive. Moving to unexpected places will likely destroy the sample and could also damage the objective or stage. Having an open text box could leave an accidental mouse click an keyboard press with very bad consequences.

carandraug, 2020-05-19:
>> Which brings another idea (for later), of a control to just type the coordinates. In the mean time, I will try a disabled TextCtrl and see what we get.
>
> This functionality is in a right click menu at the moment.

I didn't know about that. This is not very easy to discover.

> I think having ti front and central isn't a great idea as it could easily be very destructive. Moving to unexpected places will likely destroy the sample and could also damage the objective or [...]

It's exactly to prevent those accidents that we have `cockpit.gui.safeControls` (see #422). If we were to move that functionality into the window in that manner, we wouldn't take up extra space while also making the functionality easy to discover.

carandraug, 2020-05-20:
>> How about putting the positions (eg XY and Z) above the relevant sections, then we dont take up any more space but key people into what each section might represent?
>
> Sounds good to me.

Turns out that this is not simply moving the positions text up, because of the weird layout of this window. If we just move the axes position text to the top, the rows under the macro stage will still take up the same space because of the height of the saved top and bottom position. Here's the current, approximate, layout:

```
 0     1     2     3     4     5     6     7     8     9     0     1
 |------------------------------------------------------------------
 |                       |     |                             |     |
 |                       |     |                             |     |
1|                       |     |                             |     |
 |                       |     |                             |     |
 |                       |     |                             |     |
2|                       |     |                             |     |
 |                       |  E  |                             |  E  |
 |     MacroStageXY      |  M  |                             |  M  |
3|                       |  P  |           MacroStageZ       |  P  |
 |                       |  T  |                             |  T  |
 |                       |  Y  |                             |  Y  |
4|                       |     |                             |     |
 |                       |     |                             |     |
 |                       |     |                             |     |
5|                       |     |                             |     |
 |                       |     |                             |     |
 |                       |     |                             |     |
6|---------------------- |     |------------------------------------
 |     XY AxesPosition   |     |  Z AxesPosition |                 |
 |                       |     |                 |       Save      |
7|-----------------------|     |---------------- |     TopBottom   |
 |       XY buttons      |     |    Z buttons    |       Panel     |
 |                       |     |                 |                 |
8|------------------------------------------------------------------
```

But maybe we don't the labels at the top, their shape is enough to identify them? Also, I think this needs a bit of re-organization so that:

* accomodate being able to resize it and have the macro stages canvas expand with more space (to better user @dstoychev work on #587)
* the buttons are better placed. Of the three XY buttons, "Recenter" is actually meant for the Z axis, and while "Switch control" in theory works for all, I believe we mostly mean it to be used for Z.
* the hardcoded, fixed sizes of the save top bottom panel don't fit the whole text (or at least they don't on my system).

Anyway, I have made some changes according to @dstoychev suggestions, so it looks like this (I haven't bothered with borders and alignment yet, if we're going to relayout the window):

[image]

I disable/enable the text control to show which stage handler is active but even the active control is read only. This is on my `wip-594-macrostage-relayout` branch.

iandobbie, 2020-05-21:
> * the buttons are better placed. Of the three XY buttons, "Recenter" is actually meant for the Z axis, and while "Switch control" in theory works for all, I believe we mostly mean it to be used for Z.

This is not strictly true. Recentre actually recentres all axis. The original OMX and V2 actually have piezo on X, Y and Z and so might need this functionality on all axis. Current systems seem to have finer control on Z compared to XY and so associating it with Z makes more sense but I can imaging light sheet systems with rotation axes etc where recentre might apply to arbitrary axis. 

I dont mind moving the control but we should not remove the possibility of it apply to multiple axes.

carandraug, 2020-05-21:
>> * the buttons are better placed. Of the three XY buttons, "Recenter" is actually meant for the Z axis, and while "Switch control" in theory works for all, I believe we mostly mean it to be used for Z.
>
> This is not strictly true. Recentre actually recentres all axis. [...]

Ok. But the point still stands. Two of the three XY buttons are actually XYZ and so should not be there.

carandraug, 2020-06-17:
This issue has moved away from the title of the window itself, to replacing the drawing of the stage position with ftgl with wx controls and the changes to the layout that it requires so I renamed the issue.

Anyway, here's my current attempt at laying out the window.

[image]

It merges the xy and the z position text and aligns it with the save top bottom panel. I also put the row of buttons in the centre closer to the macro stage panels since I expect users to be interacting with those more while the positions table is more for read-only. But moving them is trivial and this is how they would look at the bottom:

[image]

You can test this on my `wip-594-macrostage-relayout` branch.

iandobbie, 2020-06-17:
What is the indication of which stage is being controlled? This was indicated by emphasis colour on the multiple positions, but it is not at all clear in this layout how you would do this?

carandraug, 2020-06-17:
> What is the indication of which stage is being controlled? This was indicated by emphasis colour on the multiple positions, but it is not at all clear in this layout how you would do this?

It's the one that has the textfield active (in GTK3, the background is white, while disabled ones are grey).

iandobbie, 2020-06-17:
But then the actual xy position is not displayed anywhere.

carandraug, 2020-06-17:
> But then the actual xy position is not displayed anywhere.

We discussed this in person. It never was. The macro stage never displayed the real XY position, only the position on each handler. The change should not be removing any information from the window.

carandraug, 2020-06-17:
After speaking with @iandobbie in person, I have merged the proposed changes. Closing as fixed.

closed 2020-06-17 by carandraug