microscope-cockpit/cockpit #597: Suggested changes to light path functionality.

iandobbie, 2020-05-05:
I fixed the issues and pull changes in #596.
There are a couple of outstanding suggestions not todo with the low level functionality. 
1) change the name to "Channel \n config", which then chnages to "Channel \n DAPI" or whatever channel name you chose. 

2) Change the storage a retrieval slightly. If we have a menu bar we could have the main commands in this and the button just allows access to the already created channels. The channels could be saved into peoples config files so they get back the same channels the next time they start up that instrument. We would need these menu functions:

New - create a new channel with the current config and name it
Delete - remove channels from list and config file.
Export - save to disk outside config file
Import - open from disk outside config file

carandraug, 2020-05-08:
I'm looking into this and I'm wondering if we should be doing some validation of the loaded files.

1. if we load a file that tries to set devices that do not currently, should we error?
2. and the opposite, if we load a file that does not include devices that do exist, should we error?

I think we should error in both cases above since the config file is obviously not adjusted to the current setup and we may end up setting everything incorrectly. @iandobbie and @juliomateoslangerak  what do you think?

carandraug, 2020-05-09:
I have implemented the menu as suggested and placed it on the menu bar. Changing between configurations is still the same number of button clicks as before (click on the menu, and then on the choice of configuration). I have also experimented with adding a list of buttons on the main window so you can change between configurations with only one button click. Here's a screenshot:

[image]

It's still missing a lot of plumbing to make it work, this is mostly a mockup available on my `wip-597-light-path-menu` branch.

iandobbie, 2020-05-11:
Looks good. I would suggest "channels" rather than exposure. Should be simple to plumb in you only need to call the existing functions, maybe rename them if we change the section name to make the function consistent with the section name.

carandraug, 2020-05-12:
> Looks good. I would suggest "channels" rather than exposure.

OK.

> Should be simple to plumb in you only need to call the existing functions, maybe rename them if we change the section name to make the function consistent with the section name.

It's actually a bit more involved because now we have two places showing a view on the list of channels. One is on the main window panel and the other is on a menu of the menu bar of the main window. We don't want to have to keep two lists of channels. I'm thinking this should go somewhere on the imager interface, kinda like the saved locations on the stage go into the stageMover interface.

Anyway, on a first step, I'll only add the channels menu bar to keep changes to minimum.

iandobbie, 2020-05-12:
Ok, if you add the menu bar, then I suggest you delete the options for new, load, save etc from the selection button.

carandraug, 2020-05-12:
I now made the suggested changes to the buttons and moved them all to the menu bar. Things still missing are:

1. validation of the imported channel settings
2. move the logic of keeping a list of channel settings somewhere else (possibly the imager interface)
3. add one click away button on the main window panel (on the mock above I used a WrapSizer)
4. tests

Testing the GUI is still tricky and I didn't want to be writing tests if later the logic is changed out of the GUI into the imager interface. Here's 7 manual tests to check that all works:

1. Adding a new channels add a new item on the menu.
2. Adding an existing channel will prompt for confirmation to replace it.
3. Removing a channel removes its menu
4. Export creates a file with all channels and import on a new session imports them all back.
6. importing a file in a session with existing channels, merges the two.
7. importing a file with duplicated channels, warns about it and replaces it.
8. Importing a file with duplicated channels, will do nothing if after the warning it is cancelled.

iandobbie, 2020-05-13:
Nice thought about overwriting existing channels, I had not considered this.

iandobbie, 2020-05-13:
I have a strange issue with this new channels control. Enabled cameras dont appear to be completely enabled. They are on in the mainWindow but a snap command from the snap-image button does not produce an image on them. Clicking them on and off in the mainwindow  makes them work properly. 

Create one channel with 1 camera active. - a
create a second channel with 3 cameras active. - b
test snap - all 3 cameras work.
active channel a 
Snap works on one camera.
activate channel b
2 of the 3 cameras work, one doesn't work. click to disable and re-enable that camera and it starts to work properly.

carandraug, 2020-05-13:
> I have a strange issue with this new channels control. Enabled cameras dont appear to be completely enabled. 

This sounds similar to @juliomateoslangerak issue #510 from last year and I can reproduce this. The main difference from issue #510 is that there the effect was the enabled/disabled button on the main panel while in here we see the effect on the camera view panel.

I will investigate if the issue is that the cameras are really not being enabled or the problem is on the camera view panel.

carandraug, 2020-05-13:
Looking deeper into this, I have found:

* the issue was still present back in 18ad34204125f1936c951748 when there was only a light path button on the main panel so the issue is independent of the new changes to the light path
* adding a print statement on the `MicroscopeCamera.softTrigger` I see that the cameras have been properly enabled and do take an image

So my guess is that the issue is somewhere else. I have also found that sometimes it works fine and that sometimes I also get this error:

```
Error in subscribed callable cockpit.interfaces.imager.<lambda>().  Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/events.py", line 94, in publish
    func(*args, **kwargs)
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 111, in <lambda>
    events.subscribe(events.CAMERA_ENABLE, lambda *args: self.updateExposureTime())
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 79, in wrapper
    result = func(*args, **kwargs)
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 121, in updateExposureTime
    [c.setExposureTime(e_max) for c in self.activeCameras]
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 121, in <listcomp>
    [c.setExposureTime(e_max) for c in self.activeCameras]
RuntimeError: Set changed size during iteration
```

carandraug, 2020-05-13:
I'm pretty sure this is a threading issue and unrelated to the rest of the changes done as part of this issue so I opened issue #603 for it.

iandobbie, 2020-06-01:
I think keeping a dedicated button for this is that this improves discoverability. Having the channels control in the menu bar doesn't make it easy to find. I know there are no more clicks, but part of the gui is to put frequently used functions, front and centre.

carandraug, 2020-06-04:
I experimented with creating a new interface named Channels which is effectively a list of the channels, and added a button on the main panel like the mock above. Can someone test my `wip-597-exposure-buttons` branch?

Things that I think need some thought:

1. the main window panel looks odd empty until some channels have been added
2. maybe we should have a config entry for files with default channels to load? Or should it be the same list as previously used?
3. I'm still unsure of the right sizer flags to get the buttons nicely aligned. It seems to work on GTK3 but after adding too many channels the main window needs to be adjusted manually for new buttons to appear.

iandobbie, 2020-06-05:
This seems pretty good I think. You could populate the list with a button "None" or similar at the start. 

Not sure about saving and restoring default channels. I can see arguments for use config, system config or a system default which is overridden by user channels. 

Sizers I dont know. we could have it switch to a scrolled list once it got longer than N

iandobbie, 2020-06-08:
Sorry I didn't make it clear above. I have tested this on mac and it works well. We could merge this and close the issue, maybe open a new issue about saving and restoring channels on close and open. If we should have default system channels, and how default channels interact with user defined channels.

carandraug, 2020-06-08:
I have rebased and pushed the proposed changes then.

> This seems pretty good I think. You could populate the list with a button "None" or similar at the start.

I think it would be better to simply have this panel hidden when there are no channels. That's what we do with the `FilterControls` panel. However, I'm having issues to get that right. If cockpit starts without any channels, then it would start with the panel hidden and I can't get wx to do the right in that case.

But if this channels stuff is such a central cockpit feature, then we should have a way to have users

> Not sure about saving and restoring default channels. I can see arguments for use config, system config or a system default which is overridden by user channels.

I propose the following: default does nothing, then on config files there is a `channel-files` entry on the `global` section. If that is defined on the system config file, then the user can override it on its own config file, like all other stuff on the `cockpit.conf` file.

> Sizers I dont know. we could have it switch to a scrolled list once it got longer than N

The `wx.WrappedSizer` seems to work here. I'm just wondering if it works well on other platforms.

carandraug, 2020-06-08:
> I propose the following: default does nothing, then on config files there is a channel-files entry on the global section. If that is defined on the system config file, then the user can override it [...]

I have a proposed implementation of this, with documentation included, on my `wip-597-channel-files` branch.

iandobbie, 2020-06-15:
Just tested on a mac, rebased against current master as I need the log window fix. 

Seems to work well overall. One issue is if you assign channels one by one then the buttons in the channels panel are stacked vertically. However when loaded from a config file they are laid out horizontally.

I have 3 loaded channels which are in a 3 across display. Adding a fourth makes the panel be 2x2 buttons, but it is resized so it is only 1 and a bit button wide so half the buttons aren't visible.

carandraug, 2020-06-15:
> [...] One issue is if you assign channels one by one then the buttons in the channels panel are stacked vertically. However when loaded from a config file they are laid out horizontally.

This has to do with the sizer issues I mentioned earlier. I can't get the right sizer flags and calls to `Layout` and `Fit`, and I'm not sure if the flags are wrong of this is because of the rest of the window.

carandraug, 2020-06-16:
@iandobbie can you try my `wip-597-channel-files` branch again? I have already rebased it and force pushed the branch so should work fine for you on Mac.

iandobbie, 2020-06-17:
There are two problems I see with this. 

1) The button sizer still doesn't work and it is possible to make buttons off the panel which you then can't access.

2) removing a channel deletes the menu entries but doesn't remove the button.

carandraug, 2020-06-17:
>     1. The button sizer still doesn't work and it is possible to make buttons off the panel which you then can't access.

How? It seems to work fine here, and the window is resized as needed when new channels are added.
 
>     2. removing a channel deletes the menu entries but doesn't remove the button.

Thanks. Pushed a new commit and should now be fixed.

carandraug, 2020-07-07:
This seems to be working on Windows and GTK3 so I pushed the commits that add channel files configuration option as well as other tweaks to support it. Closing as fixed. I will open a separate issue to discuss validation of channels.

closed 2020-07-07 by carandraug