microscope-cockpit/cockpit #599: Show/hide windows functionality slightly broken.

iandobbie, 2020-05-08:
The show/hide and raise to top functionality need to be connected. If you hide the python shell then raise to top you get a blank gray window. We need to also show the window if raise to top is called on a hidden window.

carandraug, 2020-05-08:
All of that seems to work for me. Can you give the exact steps to try and reproduce on my side?

iandobbie, 2020-05-08:
Yes, open the python window and then hide it. Next raise to top and you get...
[image]
Then do show/hide to get
[image]

If hidden but raise to top on the mac you get a window but no useable panel in it.

carandraug, 2020-05-08:
I can't reproduce it with wx version 4.0.7.post2 gtk3 (phoenix) wxWidgets 3.0.5. I'm doing:

1. start cockpit with `cockpit --no-config-files` (there is no PyShell window by default)
2. right click on the main window to get the window menu, then "PyShell > Show/Hide" (the PyShell window appears)
3. same as above, the PyShell window disappears
4. similar as above but "PyShell > Raise to top", the PyShell window appears correctly

iandobbie, 2020-05-11:
Ok seems to be either wx version or Mac specific. I am on wx version 4.0.6, not sure about wxWidgets version number. I think a simple test to see if the window is hidden and call window.show() if hidden would solve this. I will try it on my mac.

carandraug, 2020-05-11:
> I am on wx version 4.0.6, not sure about wxWidgets version number.

On the Python shell while running cockpit, run `import wx; print(wx.version())`

> I think a simple test to see if the window is hidden and call window.show() if hidden would solve this.

Just to be sure, does this happen with all windows or only with the Python Shell? I would like to report this upstream as well. Can you check what behaviour do you get with this?

```
import wx

class TestFrame(wx.Frame):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        panel = wx.Panel(self)
        label = wx.StaticText(panel, label=wx.version())
        sizer = wx.BoxSizer()
        sizer.Add(label)
        panel.SetSizer(sizer)

app = wx.App()

frame = TestFrame(None)
frame.Show()
frame.Hide()
frame.Raise()
app.MainLoop()
```

Looking into the recent changes, I think the reason this issue did not exist before is that the secondary windows, the ones who could be hidden, did not had a 'Raise to top' menu option.

iandobbie, 2020-05-11:
This give me a blank Window, whereas if remove the frame.hide and frame.raise I get the expected wx version info.

iandobbie, 2020-05-11:
I have a fix that just does a w.Show() followed by the w.Raise() call on selecting this menu entry. 
https://github.com/iandobbie/cockpit/tree/raise-fix

carandraug, 2020-05-11:
I have pushed your fix after editing the commit message to point to this issue and adding a comment linking to the upstream issue I just opened https://trac.wxwidgets.org/ticket/18762

Closing as fixed.

closed 2020-05-11 by carandraug