microscope-cockpit/cockpit #600: zStackMulti experiment is "hacked all to hell"

carandraug, 2020-05-08:
From the very comments on "cockpit/experiment/zStackMulti.py":

```
## Just like a standard Z-stack experiment, but we do multiple exposure times
# at each Z slice. Hacked all to hell though since we rely on OMXT's
# delay generator being set up for us ahead of time.
```

In addition to the comment, I just found out that it has hardcoded handler names:

```
        shutter = depot.getHandlerWithName('488 shutter')
        delayGen = depot.getHandlerWithName('Delay generator trigger')
```

And is not even listed on the experiment registry.

Is anyone attempting to fix it or should we just remove it?

iandobbie, 2020-05-11:
I think we need to have this functionality, especially to sell the AI microscopy idea as we need to be able to load a series of positions and run an experiment over them. Is it easier to fix this or start from scratch I'm not sure.

carandraug, 2020-05-11:
> I think we need to have this functionality, especially to sell the AI microscopy idea as we need to be able to load a series of positions and run an experiment over them. Is it easier to fix this or start from scratch I'm not sure.

This is not about a z stack experiment on multiple locations. This is about taking multi z stack images on the same location, but changing the exposure settings each time, so I don't think we will be using this for the AI microscopy idea.

closed 2020-05-17 by carandraug