microscope-cockpit/cockpit #602: camera view window fails unless the mosaic view has been shown

carandraug, 2020-05-13:
If the mosaic window has not yet been show, then the camera view also fails to displays taken images. To reproduce:

1. start cockpit and close all windows except the camera view
2. close cockpit and start again (only the camera view window will be shown)
3. enable one camera and light source
4. take an image (press snap on the main panel)
5. the logging panel will pop with this error:

```
Error drawing view canvas: cannot convert float infinity to integer
Error drawing view canvas: cannot convert float infinity to integer
```

and this warning

```
/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py:228: RuntimeWarning: divide by zero encountered in true_divide
  nx = int(np.ceil(data.shape[1] / self._maxTexEdge))
  File "/home/carandraug/.local/bin/cockpit", line 11, in <module>
    load_entry_point('cockpit', 'gui_scripts', 'cockpit')()
  File "/home/carandraug/src/cockpit/cockpit/__init__.py", line 312, in main
    app.MainLoop()
  File "/home/carandraug/.local/lib/python3.7/site-packages/wx/core.py", line 2167, in MainLoop
    rv = wx.PyApp.MainLoop(self)
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 653, in onPaint
    traceback.print_stack()
  File "/home/carandraug/.local/bin/cockpit", line 11, in <module>
    load_entry_point('cockpit', 'gui_scripts', 'cockpit')()
  File "/home/carandraug/src/cockpit/cockpit/__init__.py", line 312, in main
    app.MainLoop()
  File "/home/carandraug/.local/lib/python3.7/site-packages/wx/core.py", line 2167, in MainLoop
    rv = wx.PyApp.MainLoop(self)
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 653, in onPaint
    traceback.print_stack()
  File "/home/carandraug/.local/bin/cockpit", line 11, in <module>
    load_entry_point('cockpit', 'gui_scripts', 'cockpit')()
  File "/home/carandraug/src/cockpit/cockpit/__init__.py", line 312, in main
    app.MainLoop()
  File "/home/carandraug/.local/lib/python3.7/site-packages/wx/core.py", line 2167, in MainLoop
    rv = wx.PyApp.MainLoop(self)
  File "/home/carandraug/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py", line 653, in onPaint
    traceback.print_stack()
```

Note that the mosaic view does not have to be shown for the camera view, it only needs to have been shown before. For example, if after start cockpit we show and then close the mosaic view, then the camera view will work fine.

iandobbie, 2020-05-14:
I suspect that this is about setting the min/max display values as the masaic uses the values from the viewcanvas.

carandraug, 2020-06-29:
This happens because `viewCanvas.Image._maxTexEdge` (which is `glGetInteger(GL_MAX_TEXTURE_SIZE)`) is set during `Image` construction.  Because this happens during construction, `viewCanvas.ViewCanvas.SetCurrent` has not been called yet which is `GL_MAX_TEXTURE_SIZE` is zero.

The reason why this works if the macro stage has been shown before, is that it's enough to have set any context to current to get `GL_MAX_TEXTURE_SIZE`. So if the macro stage, touchscreen, or mosaic window (all the windows with `GLCanvas`) have already been shown, then the camera view will work fine. But windows should be independent of each other so this shold still be fixed.

carandraug, 2020-06-29:
Now fixed by moving the get of `GL_MAX_TEXTURE_SIZE` until `Image` is being drawn. I think we could avoid that by creating a new `Image` instance each time there is a new image (instead of reusing the same `Image` instance and causing it to update). I'm not sure what was the reason for that design but it would require changing too much stuff on the viewCanvas so I shied away from it.

closed 2020-06-29 by carandraug