microscope-cockpit/cockpit #603: threading issue when enabling multiple cameras

carandraug, 2020-05-13:
This issue was originally reported by @iandobbie as part of the discussion of issue #597 

The `Imager` class subscribes twice to the camera enable event. One of them modifies the set `Imager.activeCameras` while the other modifies that set. When there's multiple cameras being enabled at the same time, as is the case with the channels/LightPath buttons, this set sometimes is being modified while on another thread is iterating through it. The error caused by this issue is the following:

```
Error in subscribed callable cockpit.interfaces.imager.<lambda>().  Traceback (most recent call last):
  File "/home/carandraug/src/cockpit/cockpit/events.py", line 94, in publish
    func(*args, **kwargs)
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 111, in <lambda>
    events.subscribe(events.CAMERA_ENABLE, lambda *args: self.updateExposureTime())
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 79, in wrapper
    result = func(*args, **kwargs)
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 121, in updateExposureTime
    [c.setExposureTime(e_max) for c in self.activeCameras]
  File "/home/carandraug/src/cockpit/cockpit/interfaces/imager.py", line 121, in <listcomp>
    [c.setExposureTime(e_max) for c in self.activeCameras]
RuntimeError: Set changed size during iteration
```

This can be reproduced _sometimes_ with the following steps :

1. start cockpit with `cockpit --no-config-files` which will create 4 dummy cameras
2. enable one light source
3. with all cameras disabled 'Channels > Add channel..." with name "foo"
4. enable all four cameras and then 'Channels > Add channel..." with name "bar"
5. select "Channels > foo" and then "Channels > bar" until the error happens.

carandraug, 2020-05-13:
It is possible that this is related to, and maybe even the same as, issue #510 .

iandobbie, 2022-07-14:
Using the method suggested above to reproduce this it happens basically every time I switch between all simulated cameras on and all cameras off. 

Unfortunately my suggested fix of removing the second subscription to the ENABLE event and simply calling the updateExposureTime() at the end of the toggle event doesn't actually fix the issue. I see this as a GUI threading issue, so we should probably use the cockpit.gui.EvtEmitter as suggested in the recent fix for issue #817. I will try this.

iandobbie, 2022-07-14:
This doesnt work as it produces the following error:

Traceback (most recent call last):
  File "/Users/ID/src/cockpit/cockpit/__init__.py", line 156, in OnInit
    self._imager = cockpit.interfaces.imager.Imager(
  File "/Users/ID/src/cockpit/cockpit/interfaces/imager.py", line 97, in __init__
    cockpit.gui.EvtEmitter(self, events.CAMERA_ENABLE).Bind(
  File "/Users/ID/src/cockpit/cockpit/gui/__init__.py", line 118, in __init__
    assert isinstance(parent, wx.Window)
AssertionError

Not sure why adding the EvtEmitter call breaks the assertion that the parent is a wx.Window.

iandobbie, 2022-07-14:
Using the decorator (below, my branch https://github.com/iandobbie/cockpit/tree/cam_enable_race) to force the toggle to be called in  the main thread completely eliminates the issues in my testing with simulated devices. The start of the toggle function becomes:

    ## Add or remove the provided object from the specified set.
    @cockpit.util.threads.callInMainThread
    def toggle(self, container, thing, shouldAdd):

I totally concede David's point (see issue #817) that this should probably use the gui.EvtEmitter functions but I couldn't get that to work, see debug trace above. Open to better solutions. 

My

carandraug, 2022-07-15:
> I see this as a GUI threading issue

It's a threading issue but one a GUI threading issue. We just need to ensure that the threads run over each other, no need to force it to run on the main GUI thread (which the `EvtEmitter` and `callInMainThread` do as they're solutions for the GUI part of cockpit).

> Not sure why adding the EvtEmitter call breaks the assertion that the parent is a wx.Window.

The `Imager` is not a `wx.Window`.

----

I have pushed c4ec9abef382cde8ce51ccc6 which uses a threading lock to ensure that we don't add/remove stuff to the set of active lights/cameras and change settings on those items at the same time. Closing as fixed.

carandraug, 2022-07-15:
It has occurred to me that while this fixes the issue when adding/removing channels, it does not fix the issue when changing exposure times. To reproduce:

1. start cockpit with cockpit --no-config-files which will create 4 dummy cameras
2. enable all light sources with exposure time set to 100 msec (default)
3. enable camera 1 and 2
4.  save channel with 'Channels > Add channel..." with name "foo"
5. change exposure time for all light sourcs to 200 msec
6. disable camera 1 and 2 and enable camera 3 and 4
7. save channel with 'Channels > Add channel..." with name "bar"
8. change between foo and bar channel until the error happens.

carandraug, 2022-07-15:
I have pushed 262339524ae6 which adds the same lock logic to the update exposure time event handler. Closing.

closed 2022-07-15 by carandraug