microscope-cockpit/cockpit #605: primitives should not be a PositionerHandler property

carandraug, 2020-05-14:
A device stage creates a `PositionerHandler` for each of its axis. One of the stage device configurations are the primitives (list of shapes to draw on the stage). The GUI then gets these primitives from the `stageMover`, which gets them from all the `PositionerHandler` instead of the actual stage. This is odd because the primitives should be the same for all handlers, or at least the same for all handlers of the same device.

I'm wondering if it makes sense to move this from the devices configuration to cockpit configuration.  We even have a stage section on the cockpit config files (currently used for `dishAltitude`, `slideAltitude`, `slideTouchdownAltitude`, `loadPosition`, and `unloadPosition`).

carandraug, 2020-05-14:
Proposed fix on my `wip-605-primitives-in-cockpit-conf` branch. There's a few FIXME that I will address if people agree with the proposal. The only thing that will be missing is the ability for the linkam stage to not define a `primitives` value and instead define a `xoffset` and `yoffset`.

iandobbie, 2020-05-15:
I don't understand the point about the linkam stage. I also worry a bit about proliferation of the config files, especially if they are in different directories. There is the info about load/unload and dishaltitude etc but these seem to be mostly atrophied.

carandraug, 2020-05-17:
> I don't understand the point about the linkam stage.

All stages can have a `primitives` value on their config. The linkam stage can, instead of a `primitives` values, have an `xoffset` and `yoffset` which will draw a circle at he given offset and two others 4000 away each way on the x axis, all with a radius of 1500. Basically, if the stage has dimensions `((0, 20000), (0, 20000))` , these two are equivalent:

```
xoffset: 10000
yoffset: 10000
# if primitives is defined, then xoffset and yoffset are ignored
primitives: c  6000 10000 1500
            c 10000 10000 1500
            c 14000 10000 1500
```

> I also worry a bit about proliferation of the config files, especially if they are in different directories.

I'm not proposing to add another config file. I'm moving the configuration that draws shapes on the macro stage from the device configuration file (`depot.conf`), to the cockpit configuration. The reasoning is that having the `primitives` values on the individual axis objects, which is where they effectively end up at the moment, is weird. The primitives value is dependent on what is loaded on the stage, not a configuration of the stage device itself. In the same manner that we don't have the dish altitude on the device configuration, the shapes to draw should also not be there.

I might be seeing this wrong, but I'm seeing users selecting different primitives based on what type of well plate or sample holder they are currently using.

carandraug, 2020-05-20:
I spoke with @iandobbie earlier today and we agreed on making this change and adding an explanation of the config to the documentation. Closing as fixed.

closed 2020-05-20 by carandraug