microscope-cockpit/cockpit #606: Possibility to enlarge Image (camera) window in cockpit

VroniPfann, 2020-05-16:
When testing Cockpit on our light sheet system, I noticed that one cannot enlarge the Image (camera) window, which is quite unpractical for our application. I mentioned this already briefly to David, I just thought to also mention it here. Perhaps, you will find some time to have a look at this.

carandraug, 2020-05-17:
We have recently change this so one now can resize the camera view window. However, the window content does not expand with the window which needs to be fixed

iandobbie, 2022-06-08:
I have a mostly working implementation but it has to jump through a few hoops. Check out my branch
https://github.com/iandobbie/cockpit/tree/camera-resize

Leverages the Ysize and effectively ignores Xsize.
Header and histogram stay the same size, they should scale
Once you resize the window you seem to need a wx context switch to be able to right click in window. A bit strange

I'm sure there are other issues too.

iandobbie, 2022-06-08:
Done a bit more optimization so it get the correct size at the start etc. A few outstanding problems:

1) Once you resize the left and right click mouse actions don't work. Snap and new image and they recover

2) The forth active camera (I have 4 cameras, so maybe the last) loses its histogram until you resize, then it appears. 

3) historgram and camera header should probably scale, but not essential. 

4) only tested on macos so far.