microscope-cockpit/cockpit #608: change primitives while cockpit is running

carandraug, 2020-05-20:
While fixing #605 I found that there is code that, each time it draws the macro stage and mosaic, checks if the primitives have changed. However, there is no way for a user to actually change them during a session.

I thought of just removing it but maybe this could be a nice new feature. The use case I imagine is a user swapping plate during one session. What do you guys think? Remove or improve?

mickp, 2020-05-20:
I think that was exactly the use case it was put in for: changing the
sample holder type. I can't remember the code, but I expect originally it
evaluated the primitives on every call, then I put in a check that uses a
shortcut if they hadn't changed.

iandobbie, 2020-05-21:
I would head for improving if at all possible, or leave as a check for later improvement. Fixing something like this to require a restart it silly.

carandraug, 2020-05-21:
> Fixing something like this to require a restart it silly.

I'm not fixing this to require a restart. Requiring a restart _and_ change of configuration file is the current state of affairs, even before my changes on #605. I think we should have a way to do this without requiring that.

But it seems that no one is doing this any time soon anyway so I have removed those checks with 1ee2ad710122. Now that we have an issue about this, when someone wants to implement it, they can easily revert those changes. However, when this does get implemented, I think that instead of a check for new primitives on each paint, the canvas needs to be notified when the primitives are changed. The canvas them updates the list that is available during paint.