microscope-cockpit/cockpit #609: remove the ability to not draw primitives

carandraug, 2020-05-21:
On the code comments about a needed refactor of `MosaicCommon` I found the following entry from 3b536658bc08:

```
    #   eliminate self.drawPrimitives - they're useful for navigation and don't add much clutter.
```

I can easily remove this if no one opposes and I guess it will make any future refactor simpler.

carandraug, 2020-05-21:
Proposed code to remove this on my `609-remove-no-draw-primitives` branch.

iandobbie, 2020-05-21:
I don't quite understand. Are you just removing to ability to turn the primitives on and off?

carandraug, 2020-05-21:
> I don't quite understand. Are you just removing to ability to turn the primitives on and off?

Yes. There was an old comment from @mickp on the code proposing that (the comment mentioned above), so here's the code to actually do it.

iandobbie, 2020-05-22:
Ok, just needed to make sure you weren't removing them all together as they are very useful.

carandraug, 2020-05-22:
I have now pushed fcdd47fd3158 . Closing as fixed.

closed 2020-05-22 by carandraug