microscope-cockpit/cockpit #612: remove setting of stage step size from the devices

carandraug, 2020-05-24:
Constructing a `StagePositioner`, which is done by the stage device, requires providing a list of step sizes (these are the distance that the stage should move when in the cockpit UI one makes 1 move in a specific direction, it is not the step size of the actual hardware) and selecting a default one.

It seems that the list of step sizes each device makes is a bit arbitrary. Currently, they are:

```
aerotech = [10, 50, 100, 500, 1000, 5000]
dummy_xy = [5, 10, 50, 100, 500, 1000]
dummy_z =  [.01, .05, .1, .5, 1, 5, 10, 50, 100, 500, 1000, 5000]
linkam_stage = [1, 2, 5, 10, 50, 100, 200]
nanomover_XY = [.1, .2, .5, 1, 2, 5, 10, 50, 100, 500, 1000]
nanomover_Z = [.01, .02, .05, .1, .2, .5, 1, 2, 5, 10, 50, 100, 500, 1000]
PhysikInstrumenteM687 = [.1, .2, .5, 1, 2, 5, 10, 50, 100, 500, 1000, 5000]
picomotor = [.1, .5, 1, 5, 10, 50, 100, 500, 1000, 5000]
```

There is also `SimplePiezo` which generates the list of step sizes dynamically and has the comment "TODO - consider moving stepSizes creation to the handler." which is what I'm proposing here.

The whole step size logic does not fit into the device class. None of the devices ever modifies them and their values are not reflected on the actual hardware, their values are arbitrary, moving them somewhere more central would mean less code on each device, and even if we really want different steps sizes between different stages then we need something different when they are all microscope remote stage devices.

The suggestion of moving the logic to the `PositionerHandler` sounds good to me at an initial stage (I think later they should be moved into the `StageMover` interface and leave handlers as the level that abstracts the difference between device classes). If others agree, then question is then what should be the default step sizes? I think we can have something doubling or halving the current step size instead of fixed step sizes.

carandraug, 2020-05-24:
There is also issue #413 which complains about step sizes, but that issue is actually three issues about the stages on DeepSIM only.

carandraug, 2020-05-29:
@iandobbie said that simply multiplying/dividing by two the previous step leads to a bad experience. Currently, the only device that automates this is `SimplePiezo` which cycles between multiplying/dividing by two or five.

iandobbie, 2020-06-05:
Carnë Draug <notifications@github.com> writes:

> @iandobbie said that simply multiplying/dividing by two the previous step leads to a bad experience.
> Currently, the only device that automates this is SimplePiezo which cycles between
> multiplying/dividing by two or five.

So this give you 1 and 2 in each order of magnitude. I think it is nicer
to have 1,2 and 5 per decade. So just 3 multipliers, 2, 2.5 and 2 then
repeat.

The other thing you need is some reasonable limits on minimum and
maximum step size.

Z step size needs to go at least down to 100 nm, xy probably safe to
stop at 1 um. Maximum might top out at 500 um in z and 5mm or 10 mm in
XY. The original OMX had xy=10*Z step size.

Ian

ARCFaria, 2020-06-11:
In order to solve this issue, one has to decide how the minimum and maximum step sizes should be defined. I summarised below the 3 options I could think of. 

Looking at ```SimplePiezo```, the min and max step sizes are either defined in the device or defined inside the function (based on the axis length). 

Initially, I was thinking about using the 2nd approach, meaning that min and max would be defined based on the axis length (option 1). However as a consequence, any XY stage (if rectangular) will have a different list of step sizes for each axis, which doesn’t seem to make a lot of sense. 

@iandobbie gave some reasonable values for XY and Z stages. We could use these values and define min and max in each device (option 2). But we would be choosing arbitrary numbers, without following a specific logic.

Another option could be to have a function that creates a list of step sizes, which would then be called in each device, to overcome the fact that the handler deals with only one axis (option 3). This function somehow would have to deal with all the axis of each device, to make sure we wouldn’t have different step sizes for X and Y - but we could still allow different steps for Z axis, for instance (not sure how such thing would look like).

Is there any preferred option? Or a different one?

iandobbie, 2020-06-12:
There needs to be some logic between XY and Z step sizes as well. If there are different numbers of steps sizes in XY and Z, then as the reach the end of the range the steps sizes can get dramatically out of sync, so 100 nm is Z and 100 m m in XY which is very unexpected. 

The original OMX software had Z step = XY step/10 which had some logic and worked pretty well IMHO, lead to fewer unexpected move sizes.

Z steps definitely need to go down to 100 nm on high res systems, but XY steps below 1 um are pretty useless. If you keep this up then what is a sensible max step size? Maybe this should be the only parameter defined by stage range?

carandraug, 2020-06-14:
> [...] what is a sensible max step size? Maybe this should be the only parameter defined by stage range?

If I got your suggestion right, then handlers would have an hardcoded min step size based on their axis (1µm for XY and 100nm for Z) and the max step would be something like 1/10 of axis length. Both of these is something that the handler knows about so the device does not need to pass it to the handler. However, as @ARCFaria mentioned, the handler for an axis does not know the length of the other axes, so to avoid having X and Y with different list of step sizes in non-square stages, the device needs to pass the max step size when constructing the handler. Is that correct? This sounds good to me.

I have also though another possibility which I think would be simpler which is handling the whole step sizes in the `StageMover` interface and instead of having a list of step sizes that we move up and down, increasing/decreasing step size would "round" to the next logical step size. Using @iandobbie suggested lists of increments, the next step size could be done by simply looking at the first digit. Something like this:

```
def increase(current_step_size):
    # FIXME: account for values less than 1 µm
    first_digit = str(current_step_size)[0]
    for s in [2 5 10]:
        if first_digit < s:
            return s
````

Another reason why I think this makes more sense is that moving by a step is an operation of higher level than what handlers are about. Handlers are the level that abstracts the individual device differences Since there is no step sizes logic in the devices, then there shouldn't be step sizes in the handlers either. Also, the `StageMover` interface is aware of all the stages and so there will be no issue with rectangular stages.

carandraug, 2020-06-18:
I have implemented the second proposal of my previous comment. Can someone try my `612-remove-step-from-handlers` branch? It removes the whole step size logic from devices and handlers and implements it on the interface with the suggested increase/decreases from @iandobbie .

The only thing that should be different, other than the new default steps sizes, is that there is no lower or upper limit. Making a too small step or too large step will behave exactly the same as if the user manually made a too small or large relative move.

ARCFaria, 2020-06-21:
When I tried your branch, it fails to initialise cockpit with the error 'SizerFlags' object has no attribute 'CentreVertical'. This happens for '4.0.7.post2 osx-cocoa (phoenix) wxWidgets 3.0.5', but not for '4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4'

```
Traceback (most recent call last):
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/__init__.py", line 168, in OnInit
    module.makeWindow(main_window)
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/gui/macroStage/macroStageWindow.py", line 327, in makeWindow
    window = MacroStageWindow(parent)
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/gui/macroStage/macroStageWindow.py", line 284, in __init__
    top_bottom_panel = SaveTopBottomPanel(self)
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/gui/macroStage/macroStageWindow.py", line 193, in __init__
    wx.SizerFlags(sizer_flags).CentreVertical().Right())
AttributeError: 'SizerFlags' object has no attribute 'CentreVertical'
```

About the implementation, I didn't completely understand the implications of not having limits for the step sizes. Is this something that should be considered as it is, or the limits are yet to be implemented? 

Without the limits, it means we can see the values growing indefinitely, right? But if the stage has a physical limit to move, why not use that as max? 

Also, I have the feeling that XY and Z stepsizes should not depend on each other all the time. I makes sense to define min step, but I feel it doesn't to move around in your sample. But I don't have any example for now to support this feeling. 

After decreasing the step sizes until reaching step sizes ~ 0, I assume the value keeps decreasing, but you can't see that happening in the interface. 
Because of that, when you start increasing them again, it takes a few steps until you start actually see the values increasing. So I think it would make sense to keep limits.

carandraug, 2020-06-22:
> When I tried your branch, it fails to initialise cockpit with the error 'SizerFlags' object has no attribute 'CentreVertical'. 

That was already on the master branch. Seems like `CentreVertical` was only added on wxWidgets 3.1.0. Fixed with 3ba8b868dcf6 .

> About the implementation, I didn't completely understand the implications of not having limits for the step sizes. Is this something that should be considered as it is, or the limits are yet to be [...]

My plan was to simply not implement them at all. It's just more code to do something that is already handled by the move methods. What value does a max step size add? I think it's only:

1. prevents having too large a step size that it would move the stage outside the limit. However, even a small step size won't help if we're near the stage limit, so having a max step size does not actually prevent trying to make a step outside the stage limits. It's the actual move methods, which the `step` method uses, that prevent the move beyond the limits, so doing this is on the step size logic is just duplicated effort that also make the step size logic more convoluted.

2. prevents from accidentally increasing the step size too much.  Unless there's a kitten on the keyboard, how much of a problem is this really? Even then, I think there would be move value by solving this case by making the step size text editable.

Am I missing any other added value to the step size limits?

> Also, I have the feeling that XY and Z stepsizes should not depend on each other all the time. I makes sense to define min step, but I feel it doesn't to move around in your sample. But I don't have any example for now to support this feeling.

I don't have a strong opinion yet. We can always lock or unlock them more later as we get user experience back.

> After decreasing the step sizes until reaching step sizes ~ 0, I assume the value keeps decreasing, but you can't see that happening in the interface.
> Because of that, when you start increasing them [...]

You don't see that on the interface because we only have two decimal places, i.e., any step size below 10nm will appear as zero. But are expecting the user to really try to get step sizes below 10nm or do we expect this to happen by accident?  If the user really wants such a small step size, then a min step size would just get on the users way so we should instead change the interface to show more decimal places or use scientific notation. If this was an accident, and if the user can't remove the kitten from the microscope room, then as mentioned on the max step size I think there's more value to make the step size text field editable.

iandobbie, 2020-06-24:
I think a sensible lower and upper limit are worth having as they stop accidental setting of the steps to either 1 pm or 4 light years. 

An editable box is nice but what do you with the step size change buttons then? Does it step to the next interval or does it apply the next multiplier to the just set step size? Seems to me an editable box is an alternative to the step size change button/key.

carandraug, 2020-06-24:
> I think a sensible lower and upper limit are worth having as they stop accidental setting of the steps to either 1 pm or 4 light years.

Yes, but does that really happen? After so many years, how many times have the upper and lower limit saved us from that? And do you agree then that having an editable text box is an alternative to prevent that issue?

> An editable box is nice but what do you with the step size change buttons then? Does it step to the next interval or does it apply the next multiplier to the just set step size?

We need to have that logic even without an editable text box. The way I implemented it was to step to the next interval (see test cases https://github.com/carandraug/cockpit/commit/2f9ab5e848085306b4c246949ca91ae31e48ab70#diff-01541c76148491e08d17d7cf557d4512R38). It is simpler (we don't have to keep track of the previous multiplier) and I believe less weird since it will keep the steps with clean roundish values.

carandraug, 2020-07-07:
I spoke with @iandobbie on Monday's meeting about this, he feels that having limits may still be useful in case a notebook falls on the keyboard or something like this. We will try with an editable step size and see what happens. Should be simple to add limits later based on some configuration (which could even default to `(0.0, math.Inf)`).

I now pushed f84cc81718004647fc46ef9a which moves the step sizes from devices and handlers into the stage mover and f84cc81718004647fc46ef9a which make the text control for the step size editable.

closed 2020-07-07 by carandraug