microscope-cockpit/cockpit #613: Cockpit fails to start on mac default python - touchscreen error

ARCFaria, 2020-05-25:
when running with default python from mac, cockpit fails to start with the following error

```
Traceback (most recent call last):
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/__init__.py", line 149, in OnInit
    module = importlib.import_module(module_name)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/gui/touchscreen.py", line 811, in <module>
    class LightToggleButton(SBitmapToggleButton):
  File "/Users/arfaria/Documents/Micron/cockpit/cockpit/gui/touchscreen.py", line 825, in LightToggleButton
    mask = wx.Mask(_bmp)
wx._core.wxAssertionError: C++ assertion "v == 0 || v == 3*0xFF" failed at /Users/robind/projects/bb2/dist-osx-py36/build/ext/wxWidgets/src/osx/core/bitmap.cpp(1563) in InitFromMonoBitmap(): Non-monochrome bitmap supplied

```

carandraug, 2020-05-25:
I have looked into this with @ARCFaria and turns out that the issue is not the default Python installation in Mac. Even after installing Python 3.7 with homebrew, the error continues.

After commenting out the touchscreen line from `cockpit/__init__.py`, the whole UI looks weird. One of the things is the abort button which does ont appear with text red. The red text comes from `wx.Button.SetLabelMarkup` which in a system without support for it defaults to the normal `wx.Button.SetLabel`.

I think this might be an issue with wxPython but I have no one idea what. @ARCFaria , if after cockpit starts you open the python shell (ctrl+P) on the main window, and do the following:

```
import wx
wx.version()
```

what do you get?

ARCFaria, 2020-05-26:
'4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4'

carandraug, 2020-05-29:
This may be related with the xWidgets or WxPython version. @iandobbie can you check if you get the same error after updating wxPython?

I checked the changelog in wxWidgets and they have made recent changes to wxMask on the Mac backend. I'm not sure which of the changes caused the issue. Ideally, we would report directly to wxWidgets but unless someone with a Mac can build the minimal C++ example, I think we better report to wxPython and let them send it upstream.

@iandobbie can you check if the following example errors for your (assuming you also get an error in cockpit after upgrading)?

```
import wx

app = wx.App()
size = 75
bmp = wx.Bitmap(size, size, depth=1)
dc = wx.MemoryDC()
dc.SetFont(wx.Font(16, wx.FONTFAMILY_DEFAULT,
                   wx.FONTSTYLE_NORMAL, wx.FONTWEIGHT_BOLD, False))
dc.SelectObject(bmp)
dc.DrawCircle(size/2, size/2, size/3)
dc.SelectObject(wx.NullBitmap)
mask = wx.Mask(bmp)

frame = wx.Frame(None)
frame.Show()
app.MainLoop()
```

If it does, can you make it an even more minimal to send upstream?

ARCFaria, 2020-05-30:
I tried it and I can reproduce this error with this minimal example. I'm using python3.7 on mac, with wxPython 4.1.0

```
import wx

app = wx.App()
size = 75
bmp = wx.Bitmap(size, size, depth=1)
dc = wx.MemoryDC()
dc.SelectObject(bmp)
dc.DrawCircle(size/2, size/2, size/3)
dc.SelectObject(wx.NullBitmap)
mask = wx.Mask(bmp)

frame = wx.Frame(None)
frame.Show()
app.MainLoop()
```

carandraug, 2020-05-31:
I have reported the issue upstream at wxWidgets/Phoenix#1654

mickp, 2020-06-02:
This one looks really familiar ... I remember hitting something like it on the mac, and it was something to do with either bitmap or device context initialisation ... something was returning None where it should have returned a handle to a DC, so the DC was getting de-allocated immediately.  Does it work if you use a BitmapDC instead of a MemoryDC?

ARCFaria, 2020-06-04:
@mickp BitmapDC doesn't exist.

mickp, 2020-06-04:
Sorry - I meant PaintDC.

ARCFaria, 2020-06-05:
> Sorry - I meant PaintDC.

It wasn't working for lack of argument. 

I've downgraded for wxPython4.0.7, and the problem went away.

carandraug, 2020-06-05:
> Sorry - I meant PaintDC.

I tried to used `wx.PaintDC` too but I'm not sure how to. `wx.PaintDC` constructor requires a window as one of its argument but there's no window at this point on time. It seems this DC is to draw on a specific windows during a paint event but the code that is causing the issue is drawing the mask for circle buttons on the touchscreen before the window even exists.

Anyway, what is the logic to deploy our own toggle bitmap instead of the default `wx.BitmapToggleButton`? We use the default toggle buttons on touchscreen for the enable/disable camera buttons?

iandobbie, 2020-06-08:
I have the same issue on upgrading the wx version '4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4'. But there are a number of other issues as well.

1) macrostage and mosaic opneGL canvas regions are all too small. Also when the turn on a camera and and light source I get stderr:

/Users/ian/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py:228: RuntimeWarning: divide by zero encountered in true_divide
  nx = int(np.ceil(data.shape[1] / self._maxTexEdge))
  File “/usr/local/bin/cockpit”, line 11, in <module>
    load_entry_point(‘cockpit’, ‘gui_scripts’, ‘cockpit’)()
  File “/Users/ian/src/cockpit/cockpit/__init__.py”, line 324, in main
    app.MainLoop()
  File “/usr/local/lib/python3.7/site-packages/wx/core.py”, line 2224, in MainLoop
    rv = wx.PyApp.MainLoop(self)
  File “/Users/ian/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py”, line 653, in onPaint
    traceback.print_stack()

and stdout says:
Error drawing view canvas: cannot convert float infinity to integer

iandobbie, 2020-06-08:
The problem is glGetInteger(GL_MAX_TEXTURE_SIZE) returns 0. I will see what it used to return.

iandobbie, 2020-06-08:
I think I have already installed a hand compiled wx versions as installing wxpython-4.0.6 gives me 

Traceback (most recent call last):
  File “/Users/ian/src/cockpit/cockpit/gui/mosaic/canvas.py”, line 348, in onPaint
    dc = wx.PaintDC(self)
wx._core.wxAssertionError: C++ assertion “window->MacGetCGContextRef() != NULL” failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py37/Phoenix/ext/wxWidgets/src/osx/carbon/dcclient.cpp(195) in wxPaintDCImpl(): using wxPaintDC without being in a native paint event
Traceback (most recent call last):
  File “/Users/ian/src/cockpit/cockpit/gui/macroStage/macroStageZ.py”, line 622, in onPaint
    dc = wx.PaintDC(self)
wx._core.wxAssertionError: C++ assertion “window->MacGetCGContextRef() != NULL” failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py37/Phoenix/ext/wxWidgets/src/osx/carbon/dcclient.cpp(195) in wxPaintDCImpl(): using wxPaintDC without being in a native paint event
Traceback (most recent call last):
  File “/Users/ian/src/cockpit/cockpit/gui/macroStage/macroStageZ.py”, line 303, in onPaint
    dc = wx.PaintDC(self)
wx._core.wxAssertionError: C++ assertion “window->MacGetCGContextRef() != NULL” failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py37/Phoenix/ext/wxWidgets/src/osx/carbon/dcclient.cpp(195) in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

iandobbie, 2020-06-08:
This goes away with version 4.0.7. I will try to download and compile from source.

carandraug, 2020-06-08:
> I have the same issue on upgrading the wx version '4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4'. But there are a number of other issues as well.

Please, let's have separate issues for separate issues. I think your issue with the mosaic and macro stage might have the same root issue as issue #602

carandraug, 2020-06-08:
> I think I have already installed a hand compiled wx versions as installing wxpython-4.0.6 gives me

The stack trace shows it was not hand compiled since the path of the build is ` /Users/robind/projects/buildbots/...` (RobinD is one of wxPython developers)

iandobbie, 2020-06-08:
> Please, let's have separate issues for separate issues. I think your issue with the mosaic and macro stage might have the same root issue as issue #602

No, the variable is the same but the cause is fundamentally different. in #602 the issue is the the variable is not initialised. Here is the problem is that the openGL.GL max texture size call returns 0. Which must be related to the wx issues as reverting to 4.0.7 fixes this. 

On a related note simply compiling wxPython and wx widgets from source doesn't help.

carandraug, 2020-06-08:
> No, the variable is the same but the cause is fundamentally different. in #602 the issue is the the variable is not initialised. Here is the problem is that the openGL.GL max texture size call [...]

OK. Still, whether that is a new issue or an existing issue, it's still a separate issue from the one here, which is the creation of button masks for the touchscreen does not work on Mac with wxPython 4.1.0.

carandraug, 2020-06-17:
From upstream:

> Under wxOSX all wxBitmaps are always 32 bpp (with or without alpha channel). Depth parameter in methods like wxBitmap::Create is ignored. Due to the antialising (which is enabled by default) even bicolour drawing creates multicolour output. If you need to enforce monochrome drawing on wxOSX you need to explicitly disable antialising for given wxDC:
> [...]
> The warning you see is raised by the assertion to check if wxBitmap used to create a mask is really monochrome, It was indeed introduced in 3.1.x and that's way you don't see it on 3.0.x.

Can someone on a Mac check if my `wip-613-disable-antialias` branch fixes this?

iandobbie, 2020-06-18:
No sorry... Error is:

Traceback (most recent call last):
  File "/Users/ian/src/cockpit/cockpit/__init__.py", line 165, in OnInit
    module = importlib.import_module(module_name)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/ian/src/cockpit/cockpit/gui/touchscreen.py", line 811, in <module>
    class LightToggleButton(SBitmapToggleButton):
  File "/Users/ian/src/cockpit/cockpit/gui/touchscreen.py", line 833, in LightToggleButton
    mask = wx.Mask(_bmp)
wx._core.wxAssertionError: C++ assertion "v == 0 || v == 3*0xFF" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/osx/core/bitmap.cpp(1563) in InitFromMonoBitmap(): Non-monochrome bitmap supplied

This is with the brew installed wxpython 4.1.0. Removing the touchscreen by commenting it out from cockpit/__init__.py things start, btut he openGL canvases seem the wrong size and we get an error on the viewcanvas when we take an image
/Users/ian/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py:228: RuntimeWarning: divide by zero encountered in true_divide
  nx = int(np.ceil(data.shape[1] / self._maxTexEdge))
  File “/usr/local/bin/cockpit”, line 11, in <module>
    load_entry_point(‘cockpit’, ‘gui_scripts’, ‘cockpit’)()
  File “/Users/ian/src/cockpit/cockpit/__init__.py”, line 333, in main
    app.MainLoop()
  File “/usr/local/lib/python3.7/site-packages/wx/core.py”, line 2224, in MainLoop
    rv = wx.PyApp.MainLoop(self)
  File “/Users/ian/src/cockpit/cockpit/gui/imageViewer/viewCanvas.py”, line 653, in onPaint
    traceback.print_stack()

The MAX_TEXTURE_SIZE returns 0 as before.

iandobbie, 2020-06-18:
Oh and the button are monochrome, but the ABORT font is larger than the other buttons which it wasn't before.

carandraug, 2020-06-22:
> No sorry... Error is:

Seems lie disabling antialiasing will not work the way because the methods to get the graphics context won't be avaiable on wxPython (see my upstream at issue https://trac.wxwidgets.org/ticket/18792 ).

As an alternative, I think we just have to manually threshold the bitmap on MacOSX. Can someone confirm that my `wip-613-force-bw` branch fixes it?

> [...] he openGL canvases seem the wrong size and we get an error on the viewcanvas when we take an image [...]
> Oh and the button are monochrome, but the ABORT font is larger than the other buttons which it wasn't before.

Those are all separate issues. The fix I'm proposing is only to fix the non-monochrome bitmaps on Mac.

carandraug, 2020-06-25:
> As an alternative, I think we just have to manually threshold the bitmap on MacOSX. Can someone confirm that my `wip-613-force-bw` branch fixes it?

This also seems to not work on OSX. Not sure why but after the drawing all the pixels on the `MemoryDC` have a value of `(-1, -1, -1, 255)`.

carandraug, 2020-06-26:
>> As an alternative, I think we just have to manually threshold the bitmap on MacOSX. Can someone confirm that my wip-613-force-bw branch fixes it?
>
> This also seems to not work on OSX. Not sure why but after the drawing all the pixels on the MemoryDC have a value of (-1, -1, -1, 255).

I have found why this didn't work. It's not mentioned on the wxPython docs but looking into wxWidgets docs for `wxDC.GetPixel` there is "This method isn't available for [...] any DC in wxOSX port and simply returns false there."

carandraug, 2020-06-26:
Here's another attempt at fixing this on Mac. Can someone on Mac try my `wip-613-force-bw-via-image` branch?

iandobbie, 2020-06-29:
This branch worked fine on wx 4.0.7 but failed on 4.1.0 with the error

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/cockpit-2.9.0+dev-py3.7.egg/cockpit/__init__.py", line 165, in OnInit
    module = importlib.import_module(module_name)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.7/site-packages/cockpit-2.9.0+dev-py3.7.egg/cockpit/gui/touchscreen.py", line 804, in <module>
    class LightToggleButton(SBitmapToggleButton):
  File "/usr/local/lib/python3.7/site-packages/cockpit-2.9.0+dev-py3.7.egg/cockpit/gui/touchscreen.py", line 818, in LightToggleButton
    mask = wx.Mask(_bmp)
wx._core.wxAssertionError: C++ assertion "v == 0 || v == 3*0xFF" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/osx/core/bitmap.cpp(1563) in InitFromMonoBitmap(): Non-monochrome bitmap supplied

carandraug, 2020-06-29:
Here's another attempt at fixing this, we manually draw the circle pixel by pixel with numpy. It works on Linux. I see no reason why it wouldn't work on OSX but I though the same on my previous attempts. @iandobbie can you give this a go? (my `wip-613-draw-circle-numpy` branch)

iandobbie, 2020-06-30:
This seems to fix the button issue, but the background is wrong. See screenshot. One of the fixes has also fixed the GL texture issue. Still a canvas size issue, I will open a new issue on it. 
[image]

carandraug, 2020-07-02:
> This seems to fix the button issue, but the background is wrong. See screenshot.

I'm out of ideas on how to fix this. Upstream suggested drawing on `wxGCDC` but I'm not sure how and after suggesting 3 fixes all of which don't work on OSX I'm getting the felling it might not work on OSX even if I figure out how to do it.

iandobbie, 2020-07-03:
I suggest we leave this issue as dormant until we get Danny's revised touchscreen code as I think this uses the native buttons. We also seem to have fixed the texture size issue as David's latest WIP-613 branch runs on my mac with wxpyhton 4.1.0.

carandraug, 2020-07-05:
> I suggest we leave this issue as dormant until we get Danny's revised touchscreen code as I think this uses the native buttons.

This issue prevents Mac devs from even starting cockpit with the latest wxPython so I think it's pretty important and Danny's revised touchscreen is an almost complete rewrite that is not ready. I found an alternative fix which changes the button colour internally sidestepping the ise of `wxMask`. @ARCFaria tested it on OSX so I pushed it (ac9fba348c5eef87c076a0).  Closing as fixed.

closed 2020-07-05 by carandraug