microscope-cockpit/cockpit #614: add support for Python microscope stages

carandraug, 2020-05-26:
Microscope now has support for stages but Cockpit still uses its own class for each stage it supports. We need a `MicroscopeStage` class similar to the `MicroscopeLaser` and `MicroscopeCamera`.

This work was already started being discussed as part of fixing #572 which is about replacing cockpit dummy devices with microscope test devices but replacing the Cockpit dummy stages with Microscope test stages requires first adding support for Microscope normal stages.

carandraug, 2020-05-26:
I have a proposal that adds support for Microscope stage on my `wip-572-microscope-stage` branch (the issue number on the branch name is because that started as fixing #572). The deviceserver configuration and `cockpit.depot` entries I'm using for testing this are:

```
from microscope.devices import device, AxisLimits
import microscope.testsuite.devices as testdevices

DEVICES = [
    device(testdevices.TestStage, '127.0.0.1', 8008,
           {'limits': {'X': AxisLimits(0, 25000), 'Y': AxisLimits(0, 12000)}}),
    device(testdevices.TestStage, '127.0.0.1', 8009,
           {'limits': {'Z': AxisLimits(0, 25000)}}),
    device(testdevices.TestStage, '127.0.0.1', 8010,
           {'limits': {'Z': AxisLimits(0, 2500)}}),
]
```

which matches the following on my depot.conf:

```
[XY stage]
type: cockpit.devices.microscopeDevice.MicroscopeStage
uri: PYRO:TestStage@127.0.0.1:8008
x-axis-name: X
y-axis-name: Y
x-units-per-µm: 1
y-units-per-µm: 1

[innerZ]
type: cockpit.devices.microscopeDevice.MicroscopeStage
uri: PYRO:TestStage@127.0.0.1:8010
z-axis-name: Z
z-units-per-µm: 10

[outerZ]
type: cockpit.devices.microscopeDevice.MicroscopeStage
uri: PYRO:TestStage@127.0.0.1:8009
z-axis-name: Z
z-units-per-µm: 1
```

There's two obvious issues remaining:

1. implementing `getMovementTime` and `cleanupAfterExperiment` (those are required to have them eligible for experiments). This is partially related to issue #496.
2. adding some sensible step sizes (actually removing them as part of #612 or some other fix for t, so I'm marking this issue as blocked). At the moment there's just hardcoded step sizes.

I'm marking this issue as blocked by #612 . I think there is also worth on considering fixing #496. Fixing #611 while definitely not require would also mean a few less lines of code required here.

iandobbie, 2020-06-12:
getMovementTime should return some vaguely sensible number. Usually this will be of the form a+b*distance, where a is some constant overhead and b*distance is the time taken to move by said distance. Seems to me this will be hardware dependant, also related to stage velocity which is settable on some stages. For the simulated stage why not make it 1ms +10ms/mm so you can test it by having large moves which might take a measurable amount of time but  we don't have to wait ages for simulated stages to pretend they are moving.

iandobbie, 2020-06-12:
cleanUpAfterExperiment I think only returns stages to pre experiment positions.

carandraug, 2020-06-12:
About movement times, the problem then is how to get that since stages in Microscope don't have that, they don't even have physical units. From what you said, I'm thinking of having two stage classes wrapping Microscope stages, something like `FixedSpeedStage` and `ModifiableSpeedStage`, the first taking a fixed value on the configuration and the later the name for a setting. I'm guessing the overhead time is always fixed.

iandobbie, 2020-06-15:
Is there any fundamental reason not to have microscope stages return  movement times? They are the hardware interfaces which is the only sensible place to calculate, or estimate these.

iandobbie, 2020-06-16:
z-units-per-µm: 1

Are we really going to have non-canonical ascii characters in our config file?

iandobbie, 2020-06-16:
The wip-572-microscope-stage stage appears to work until you try and do a mosaic. On my machine with microscope/master this produces one image every minute.
This is a serious issue as I am trying ti implement the mosaic fake stage which I think is an essential release feature to allow people to test the mosaic mode. I have no idea why this fails, especially as my test implementation did do mosaics.

iandobbie, 2020-06-16:
Looking at the code I wonder if this is a subtle change in behaviour of the events.executeAndWaitForOrTimeout call?

iandobbie, 2020-06-16:
I think the issue is the mosaic calls this...
  try:
                data, timestamp = events.executeAndWaitForOrTimeout(
                    events.NEW_IMAGE % camera.name,
                    cockpit.interfaces.imager.takeImage,
                    camera.getExposureTime()/1000 + CAMERA_TIMEOUT,
                    shouldBlock=True)

and I think the execute and wait no longer returns the values expected... I am investigating

iandobbie, 2020-06-16:
New theory. The interval changes between 30s and minute. I wonder if it is timeout on the move betwee one axis and two axis having 2 timeouts?

iandobbie, 2020-06-16:
The issue is definitely in the stagemover code. the _goToAxes function in  cockpit/interfaces/stageMover.py blocks for 30s waiting for the move to finish. I will keep trying to chase this.

iandobbie, 2020-06-16:
The microscopeDevices  Stage class never publishes a STAGE_STOPPED event, so the event is left to time out. Other stages have a sendPositionUpdates function,. which we need to add to the microscopeDevice stage classes

iandobbie, 2020-06-16:
I have a branch https://github.com/iandobbie/cockpit/tree/pos-fix that starts from your wip-572-microscope-stage and fixes this, along with another tiny issue that no-move needed would block for timeout.  It also needs an is_moving() function on the microscope side. We should discus this. Can we please move towards merging this as my mosaic branch is getting pretty removed from master.

carandraug, 2020-07-08:
Thanks for testing. I have now squashed and merged the previous proposal with the `STAGE_STOPPED` event fix and a change on the units per µm config value.

> z-units-per-µm: 1
>
> Are we really going to have non-canonical ascii characters in our config file?

I was not sure about it. I thought that a "µ" would be simple enough to type nowadays and most people copy paste config files anyway. I thought it was better than just using "u" but now I changed it to "z-units-per-micron".

The only thing missing now is to enable microscope stages to be eligible for experiments which requires implementing the `getMovementTime` callback for `PositionerHandler`:

> Is there any fundamental reason not to have microscope stages return movement times? They are the hardware interfaces which is the only sensible place to calculate, or estimate these.

The hardware interfaces might provide speed of the stage or even acceleration but not movement time. Returning an estimate of the time a movement might take is not a common function and none of the interfaces of the stages we have it provide it so I don't think it makes sense to have that in Microscope which is meant for a bit lower level.

My guess is that this is probably something that the user will need to set after calibration and so should go in configuration (see discussion on #496) . Also, this is only used for cockpit experiments because cockpit uses an action table where that time is required.

iandobbie, 2021-03-19:
The microscope stage class anc cockpit interface to it are working well, see Danny's prior and my zaber stage as well as the linkam stage. I am closing this as fixed. If we want to continue the movement time discussion it should be in a new issue.

closed 2021-03-19 by iandobbie