microscope-cockpit/cockpit #615: remove dependency on ftgl or make it easier to install

carandraug, 2020-05-28:
We are dependent on ftgl library. This has caused a number of installation issues for us, we repeatedly face the same problem (#541, #371, #346, #314, #311, #563 --- and these are only the ones we opened an issue on github). If we, who develop cockpit and know about this can't figure it out, what are the chances of new users and potential new developers?

Ideally, FTGL would not be a difficult dependency. To address this, someone would have to address the lack of binary distributions of it for non-Linux platforms (even on Mac, requiring users to install homebrew first seems crazy). I guess this could be us. We could either take on the maintenance of the abandoned pyftgl or sippy-ftgl, packages, or create our own new ftgl package.

Alternatively, we could just replace FTGL with something else.  At some point on time, I experimented with GLUT, I believe Mick did the same but FTGL gave better results. I guess we could explore it again.

To get an idea where FTGL is used:

* marked positions on the touchscreen and mosaic. Requires characters `0-9`
* histogram markers on the camera view. Requires characters `0-9[]`
* corner coordinates on macro stage. Requires characters `0-9,()`
* macro stage window coordinates labels (although it is suggested on issue #594 to replace with wx controls). Requires characters `XYZ:-0-9.stepum`

carandraug, 2020-05-28:
If I remember right, the reason to keep using FTGL instead of replacing with GLUT is that FTGL with texture fonts does the resize automatically and will always look nice. Looking into GLUT, maybe we can use its stroke fonts which should also resize automatically.

Ilan also suggested that instead of FTGL we create bitmaps of the the few characters we need at very high resolution and just use those images.

iandobbie, 2020-05-28:
Removing FTGL would definitely help the installation, but once installed it works very well and reliably.

carandraug, 2020-05-28:
On the mosaic window, this is what it currently looks like (ftgl):

[image]

I had a very quick experiment with glut and this is what I got (aliasing problems):

[image]

Another alternative is if instead of ftgl we use freetype with opengl ourselves.

carandraug, 2020-06-26:
> Another alternative is if instead of ftgl we use freetype with opengl ourselves.

We discussed in person that we would try this approach.  Effectively, this means re-implementing the bits of FTGL that we need in python (FTGL uses Freetype2 for the rendering).

I have now done this. Can someone try my `615-replace-ftgl-with-freetype` branch? Because the code is heavily based on the FTGL source, I've also added the copyright notice from FTGL.

iandobbie, 2020-06-29:
Appears to work fine on mac.

carandraug, 2020-07-05:
Nice. I have rebased and merged it. Closing as fixed.

closed 2020-07-05 by carandraug