microscope-cockpit/cockpit #618: LoggingWindow appears inside MainWindow

ARCFaria, 2020-06-05:
When I start cockpit, the loggingWindow appears inside mainwindow. I'm using Mac, python3.7 and wx '4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4'.

[image]

This happens with the commit b8462d577a838fdd94575e0150e754110ef5a309

carandraug, 2020-06-05:
This does not happen under GTK3, it may be OSX specific (I wonder what happens on Windows). The commit you mention re parents the logging window to the main window (all other windows are children of the main window). One of the things this makes work is that the logging window is also brought to the front when cockpit is activated (i.e., when we change program such as the browser, and then come back to cockpit).

What do you get with the following minimal example? I get two empty windows named Frame1 and Frame2. Do you get only one?

```
import wx

app = wx.App()

frame1 = wx.Frame(None, title='Frame1')
frame1.Show()
frame2 = wx.Frame(None, title='Frame2')
frame2.Reparent(frame1)
frame2.Show()

app.MainLoop()
```

ARCFaria, 2020-06-08:
> What do you get with the following minimal example? I get two empty windows named Frame1 and Frame2. Do you get only one?

No, I also get two windows.

dstoychev, 2020-06-08:
Same issue on Windows:

[image]

With the example above, I get only **one** window and the same sort of behaviour. Therefore, I tried to comment out https://github.com/MicronOxford/cockpit/blob/1f2ff6542c26d3a1f9538423b31bb75edbf2cb44/cockpit/__init__.py#L145 and this fixes the issue (or at least the logging window appears as I remember it before the reparenting was added).

carandraug, 2020-06-08:
With the example above, I get only one window and the same sort of behaviour. Therefore, I tried to comment out `logging_window.Reparent(main_window)` and this fixes the issue (or at least the logging window appears as I remember it before the reparenting was added).

The reparenting is needed for the logging window to behave like the others but maybe it needs to be done differently for top level windows. I don't know what and I can't reproduce the issue on GTK. I have found a related issue on wxWidgets tracker and seems we're doing it correctly. Can you check if calling `Show`after `Reparent` fixes the issue? t might be required to to do it on both or maybe in only one of them.

dstoychev, 2020-06-08:
> The reparenting is needed for the logging window to behave like the others but maybe it needs to be done differently for top level windows. I don't know what and I can't reproduce the issue on GTK. [...]

Yes, I tried your example from above and it still puts the second frame inside the first one. After a bit more playing around, I believe ``logging_window.Reparent(main_window)`` should be changed to ``main_window.AddChild(logging_window)``.

carandraug, 2020-06-08:
> [...] I believe `logging_window.Reparent(main_window)` should be changed to `main_window.AddChild(logging_window)`.

This works on GTK3 too. What's the behaviour on MacOSX @ARCFaria ? Still, I'm wary of that change since documentation recommends against its use because "Notice that this function is mostly internal to wxWidgets and shouldn't be called by the user code."

I think this is likely to be a wxWidgets bug so I reported it there at https://trac.wxwidgets.org/ticket/18785 . I will be unable to test any further changes from them. @dstoychev if you can build the minimal C++ example, could you please take the lead there?

iandobbie, 2020-06-11:
I tried Danny's suggested fix on Mac and it works fine. I think we should pull this as master is currently unusable. It may not be the right fix in the long run but it at least makes the master branch work.

carandraug, 2020-06-11:
> [...] It may not be the right fix in the long run but it at least makes the master branch work.

Yes. And upstream has confirmed the issue is on their side so there's not much we can do. I have pushed @dstoychev proposed fix in 2e9c8cbf03d2. Closing as fixed.

closed 2020-06-11 by carandraug

carandraug, 2020-07-08:
For future reference, the issue on `wxFrame.Reparent` has been fixed upstream by @dstoychev (see wxWidgets/wxWidgets@5e7e89de16bb5d184d51e2f93924a2b66f9896a1)