microscope-cockpit/cockpit #619: buttons on top of main window do not expand vertically on MacOSX

carandraug, 2020-06-05:
For whatever reason, the buttons on the top row on the main window do not expand vertically under MacOSX. This was originally reported on issue #589 by both @iandobbie and @ARCFaria under wxPython 4.0.7 and 4.1.0. See "ABORT" and "Live" button:

[image]

I confirm that under GTK3 the buttons appear expanded as expected under both wxPython 4.0.7 and 4.1.0, so this issue seems to be MacOSX specific. Not sure what happens under Windows, @NickHallONBI could you check?. Looking at the code only, I don't see anything odd that could explain it.

iandobbie, 2020-06-12:
The os buttons do not expand on macos, I have just read a stackexhange thread about it. How about just saying "Live\nvideo"? 

Other option is to move away from native buttons which seems to defeat the object of usin g wx in  the first place.

iandobbie, 2020-06-12:
On the mac adding the video give...
[image]

iandobbie, 2020-06-12:
Cant seem to do anything about the abort button. None of the markup commands seem to be able to increase the font size, and I don't think it is bold either. Couldn't find anyone else complaining about setlabelmarkup not working on a mac so I am at bit of a loss

carandraug, 2020-06-12:
> The os buttons do not expand on macos, I have just read a stackexhange thread about it. How about just saying "Live\nvideo"?
>
> Other option is to move away from native buttons which seems to defeat the object of usin g wx in the first place.

We have just replaced "Video\nmode" with "Live" because it was shorter, I don't think we should go back to something longer just to force two lines on a Mac. And I agree that moving away from the native buttons is a bad idea for the same reason as you.

I'm not sure which stack overflow thread you found. I found one that was about [setting a button size](https://stackoverflow.com/questions/29083891/wxpython-button-size-being-ignored-on-osx/29104249) and not about expand, but I think the idea is the same. Basically, we shouldn't even be trying to expand buttons on Mac because that would go against their design. So I went to take a look at the [Mac design guidelines](https://developer.apple.com/design/human-interface-guidelines/) and while nowhere they say you can't do this, in all examples, all buttons have a single line even when they were quite long. Reading from their guidelines, we should be using a toolbar instead of normal buttons, which I think makes sense (I think we've effectively been faking one anyway):

> Use a push button in a view, not a window frame. Push buttons aren’t intended for use within toolbars and status bars. If you need a button in a toolbar, use a toolbar item. See Toolbars.

I'm not sure this will solve the problem, in all their examples, even the toolbar item had a single line of text.

iandobbie, 2020-06-16:
I'm don't think "video mode" was changed to make it shorter and I think it was to make it closer to other systems. I think "Live\nvideo" is still consistent with other packages and makes the mac interface much cleaner. I think we should move to this.

carandraug, 2020-06-19:
But we are only considering changing "Live" to "Live\nvideo" to force it the same height as the others. But the text in a button should be as short as possible without sacrificing meaning so using "Live video", when "Live" can pass the same information, is not a good solution and actually makes the interface worse.

Forcing the text to be something with two lines does not address the root of the problem which is that a Mac native interface does not have extra high buttons or multi-line text on buttons. What will we do, next time we have a button there with a single word? Actually, what will we do with the "Abort" button? Shouldn't we instead be trying to make the text buttons shorter?

iandobbie, 2020-06-23:
> Forcing the text to be something with two lines does not address the root of the problem which is that a Mac native interface does not have extra high buttons or multi-line text on buttons. What [...]

Maybe so but we don't have a plan for this and it makes a mac interface look massively better. Yes the abort button is smaller but at the end that doesn't look so bad, compared to the one short button in the middle of a line of tall ones.

ChrisChong8, 2020-06-23:
By comparison, for new user, the GUI looks way cooler in Mac than Windows (no matter what changes).

mickp, 2020-06-23:
Re. note above about `SetLabelMarkup`: there's no guarantee it's supported on a given platform. From the docs:

> Simple markup supported by this function can be used to apply different fonts or colours to different parts of the control label **when supported**. If markup is not supported by the control or platform, it is simply stripped and SetLabel is used with the resulting string.

iandobbie, 2023-09-30:
This is obviously just how wx will work on Mac, no point in keeping this issue open.

closed 2023-09-30 by iandobbie