microscope-cockpit/cockpit #620: Do we need soft stage limits?

iandobbie, 2020-06-08:
As discussed in #524 we have changed the hard limit behaviour, in this issue there was some discussion of soft-limits. The soft-limits are user changeable limits which, once set, prevent the stage moving beyond these positions. 

The suggested use case is to mount a sample and then set the limit in some sample based manner, eg the extreme range of where a tissue sample is present. I have never actually used these limits and David suggested possibly getting rid of them entirely. I think we should probably keep them for the moment and see if biological users find them helpful. 

Any thoughts?

carandraug, 2020-06-08:
I too don't remember ever having set soft limits when using the microscope. Let's ask the facility if they ever use soft limits or if the users do? If no one out there uses them, and if we don't use them either, them let's get rid of them.

carandraug, 2020-07-09:
We had a discussion about this on the group meeting where Nadia and Lothar gave their opinions. We decided that we will have soft limits that can be set on the cockpit config file and defaulting to the hard limits. We will not have the ability to change soft limits at runtime.

Basically, we envision having someone (facility manager) that will set soft limits system wide to avoid having the stage move into a "dangerous" position. This still provides the ability for users to have their own local settings override the system wide settings (they may be using a special sample holder). This also provides the ability to have the facility set multiple desktop shortcuts that load different config files with command line options, for different sample holders, and the user simply restarts cockpit if they want to load different soft limits.

Lothar also asked about limiting to a specific plate to help in keeping within the area where we reasonable expect a sample to the be but that is the aim of the stage primitives and not be achieved by soft limits.