microscope-cockpit/cockpit #623: No Image Captured on Camera Views Panel (Python 3.7 in Windows)

ChrisChong8, 2020-06-12:
No noise image was captured on Camera Views panel. This was performed on Python 3.7 in Windows OS. Steps taken as follows:
- Upon starting cockpit using Git Bash, "488nm" button was pressed and followed by "south" camera button. South filter on DAPI.
- No noise image was captured when "Snap image" button was pressed.

Error message showed in logging panel:
Traceback (most recent call last):
  File "c:\users\chris\appdata\local\programs\python\python37\lib\site-packages\wx\core.py", line 3383, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "c:\src\cockpit\cockpit\gui\camera\window.py", line 97, in onCameraEnableEvent
    inactiveViews.pop().enable(camera)
  File "c:\src\cockpit\cockpit\gui\camera\viewPanel.py", line 175, in enable
    size = (VIEW_WIDTH, VIEW_HEIGHT))
  File "c:\src\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 413, in __init__
    self.image = Image()
  File "c:\src\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 164, in __init__
    self._maxTexEdge = glGetInteger(GL_MAX_TEXTURE_SIZE)
  File "c:\users\chris\appdata\local\programs\python\python37\lib\site-packages\OpenGL\latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
  File "c:\users\chris\appdata\local\programs\python\python37\lib\site-packages\OpenGL\wrapper.py", line 678, in wrapperCall
    raise err
  File "c:\users\chris\appdata\local\programs\python\python37\lib\site-packages\OpenGL\wrapper.py", line 671, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "c:\users\chris\appdata\local\programs\python\python37\lib\site-packages\OpenGL\error.py", line 234, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glGetIntegerv,
	pyArgs = (
		GL_MAX_TEXTURE_SIZE,
		<object object at 0x0000029D19D12FA0>,
	),
	cArgs = (
		GL_MAX_TEXTURE_SIZE,
		array([0], dtype=int32),
	),
	cArguments = (
		GL_MAX_TEXTURE_SIZE,
		array([0], dtype=int32),
	)
)

Thanks.

carandraug, 2020-06-12:
@iandobbie wrote:
> We have had this issue many times before. Your graphics card is reporting support for texture sizes that it can't actually deal with. Hence when the mosaic tries to create these large textures the card crashes and all the opengl display stuff fails.

Looking into our previous issues I've found #476 which is also error code 1282 when acquiring a new image. It was supposed to have been fixed with #477 (there is is also #500, #535, and #540 but those are all error code 1286 when drawing the mosaic canvas).

carandraug, 2020-06-13:
@ChrisChong8 wrote:
>  I am not sure what changes in the codes that make the camera to fail again. Before I updated my cockpit, the camera worked fine as demo.

iandobbie, 2020-06-15:
The previous issues were on the mosaic canvas where we have 

      tsize = glGetInteger(GL_MAX_TEXTURE_SIZE)
        # If we use the full texture size, it seems it's too large for manipul-
        # ation in a framebuffer:
        #   * on Macs with Intel chipsets;
        #   * on some mobile nVidia chipsets.
        # Check vendor with glGetString(GL_VENDOR)
        # GL_MAX_FRAMEBUFFER_WIDTH and _HEIGHT are not universally available,
        # so we just use a quarter of the max texture size or a reasonable
        # upper bound which has been found to work in tests on 2017-ish
        # Macbook Pro.
        tsize = min(tsize // 4, 16384)
  

However in the imageViewer/viewCanvas.py we only have
        # Maximum texture edge size
        self._maxTexEdge = glGetInteger(GL_MAX_TEXTURE_SIZE)
        
So maybe we should have the  tsize = min(tsize // 4, 16384) line in ther imageviewer module as well.

ChrisChong8, 2020-06-16:
Camera views panel works by displaying noise image upon Snap when wxPython is switched from the latest version 4.1.0 to older version 4.0.7. If wxPython is upgraded beforehand, the package has to be uninstall and to install older wxPython version. 

Steps as follows:
i) Uninstall wxPython 4.1.0
- Locate Python Scripts path i.e. C:\Users\Chris\AppData\Local\Programs\Python\Python37\Scripts
- Open Command Prompt (Run as Admin) and go the Python Scripts folder by entering i.e.
   cd C:\Users\Chris\AppData\Local\Programs\Python\Python37\Scripts
- Type in: 
   pip uninstall wxPython

ii) Install wxPython 4.0.7 (https://pypi.org/project/wxPython/4.0.7/)
- Still inside Command Prompt and under Python Scripts folder, type in:
   pip install wxPython==4.0.7

[image]

iandobbie, 2020-06-16:
You should also be able to do 

python -m pip install/uninstall ... to save navigating to the folder.

carandraug, 2020-06-29:
@ChrisChong8 can you check if this issue is still present? @dstoychev suggests that it might have been fixed while fixing issue #602 which was also on the camera view canvas but is independent of wxPython version.

carandraug, 2020-12-02:
@ChrisChong8 is this still an issue? If not, do you know what was that fixed it?

ChrisChong8, 2020-12-04:
No longer an issue. I have not done anything further since my last comment in this thread if I can recall. Thanks.

closed 2020-12-04 by carandraug