microscope-cockpit/cockpit #624: wxpython 4.1.0 issue on Mac, GL_MAX_TEXTURE_SIZE = 0.

iandobbie, 2020-06-16:
Moving from wxpython 4.0.7 to 4.1.0 on a mac changes the responce to the following call. 
glGetInteger(GL_MAX_TEXTURE_SIZE). On 4.0.7 it returns

from OpenGL.GL import *
glGetInteger(GL_MAX_TEXTURE_SIZE)
16384

on 4.1.0 this returns 0, which leads to a divide by zero error. There seems to be a fundamental opengl issue in wxpython 4.1.0

carandraug, 2020-06-16:
None of those lines call any wx code so I'm guessing this needs to be run in the Python shell from within cockpit.

carandraug, 2020-06-16:
One of the changes with wxPython 4.1.0 is that it starts to use use the development builds of wxWidgets (the 3.1.x releases instead of 3.0.x) so there are many years of changes. Still, looking at the changelogs there's not much mention of opengl in the whole 3.1 series, but there is:

> - Improve wxGLCanvas to be more type safe and better support modern OpenGL.

which could be hiding a lot of changes.

dstoychev, 2020-06-22:
> 
> 
> None of those lines call any wx code so I'm guessing this needs to be run in the Python shell from within cockpit.

You can get it to work outside the integrated shell, but it's a bit tricky. All OpenGL code requires to have a canvas setup with a context via ``SetCurrent()``. Here is the smallest working example I have:

```python
import wx
import wx.glcanvas
from OpenGL.GL import *

app = wx.App()
frame = wx.Frame(None, title="Hello World")
canvas = wx.glcanvas.GLCanvas(frame)
context = wx.glcanvas.GLContext(canvas)
canvas.SetCurrent(context)
frame.Show()

# PUT OPENGL HERE
print(glGetInteger(GL_MAX_TEXTURE_SIZE))

app.MainLoop()
```

Naturally, this prints 16384 under Windows and wxPython 4.1.0.

iandobbie, 2020-06-23:
This also works on my mac, printing 16384. I guess this means they have changed the setup required to get the openGL calls to work. Probably just requires moving around or adding to the init functions in out opengl code segments.

carandraug, 2020-06-23:
You can only call `wx.glcanvas.GLCanvas.SetContext` after the window has been show, which means after the main loop as started. With wxPython 4.1.0 gtk3 (phoenix) wxWidgets 3.1.4, the minimal example errors with:

```
Traceback (most recent call last):
    canvas.SetCurrent(context)
wx._core.wxAssertionError: C++ assertion ""xid"" failed at /tmp/pip-install-2004mlpb/wxpython/ext/wxWidgets/src/unix/glx11.cpp(584) in SetCurrent(): window must be shown
```

The following modified example works:

```
import wx
import wx.glcanvas
from OpenGL.GL import *

class OurCanvas(wx.glcanvas.GLCanvas):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._context = wx.glcanvas.GLContext(self)
        self.Bind(wx.EVT_PAINT, self.OnPaint)
    def OnPaint(self, event):
        self.SetCurrent(self._context)
        print(glGetInteger(GL_MAX_TEXTURE_SIZE))

app = wx.App()
frame = wx.Frame(None, title="Hello World")
canvas = OurCanvas(frame)
frame.Show()
app.MainLoop()
```

@iandobbie wrote:
> This also works on my mac, printing 16384.

With which version of wxPython/wxWidgets?

iandobbie, 2020-06-23:
> @iandobbie wrote:
> 
> > This also works on my mac, printing 16384.
> 
> With which version of wxPython/wxWidgets?

with wxpython 4.1.0

carandraug, 2020-06-29:
@iandobbie can you describe what's the "visible" issue here, beyond `glGetInteger(GL_MAX_TEXTURE_SIZE)` being zero? I thought that the issue was that the macrostage window looked weird. @dstoychev suggested that b9db9e0860 , which was pushed to master recently to fix a separate issue, might have fixed this too. @iandobbie , can you check?

iandobbie, 2020-06-30:
I'm not sure what you want me to check. with wxpython 4.1.0 I still appear to get the canvas about 1/4 of the expected size, but with the master branch I also cannot properly run stuff as the openGL errors mean lots of things don't work. your latest fix seemed to work but the canvas size was still wrong as I reported in issue #628

carandraug, 2020-07-02:
> I'm not sure what you want me to check. with wxpython 4.1.0 I still appear to get the canvas about 1/4 of the expected size, but with the master branch I also cannot properly run stuff as the openGL errors mean lots of things don't work.

I mean that I need more description of this issue. `GL_TEXTURE_SIZE` being zero just means that something hasn't been initialized properly yet but I have no clue what or where (I don't have a Mac to test this so I need reports). You now mention errors. Can you show those? The stacktrace might be able to help us fixing it. Also, I missed the canvas being 1/4 of the size. I thought there was also problems of sizing on the macro stage panels.

iandobbie, 2020-07-02:
Carnë Draug <notifications@github.com> writes:

> I'm not sure what you want me to check. with wxpython 4.1.0 I still appear to get the canvas
>  about 1/4 of the expected size, but with the master branch I also cannot properly run stuff as the [...]

Using your branch wip-613-draw-circle-numpy cockpit appears to work
fine under wxpython4.1.0 on the mac, snap, mosaic, touchscreen, etc. EXCEPT...

> Also, I missed the canvas being 1/4 of the size. I thought there was also
> problems of sizing on the macro stage panels.

Yes, I think all the openGL canvas's are 1/4 the size. So the
camaraviews, macrosatge XY, macrostage Z, mosaic and the touchscreen
panels are all too small and in the bottom left hand corner of the
expected place.

Ian

carandraug, 2020-07-06:
After today's meeting, @iandobbie said that this specific issue is resolved. There's open issues for the rest. Closing as fixed.

closed 2020-07-06 by carandraug