microscope-cockpit/cockpit #627: New macrostage panel z position isn't updated.

iandobbie, 2020-06-29:
The new macrostage positioning panel has a strange Z-height (um) lable next to a number that is updated in a strangely random manner. Doesn't update on Z moves. 

[image]

carandraug, 2020-06-29:
That's the height in z between the saved top and bottom positions. The label "z-height (µm):" is exactly the same as before and its position within the saved top/bottom panel is also the same. See an old screenshot:

[image]

iandobbie, 2020-06-29:
Ok. Maybe a new name. Z stack height/ size. Also maybe it should be forced positive as I assumed it was the current height and couldn’t work out what was changing it.

Ian

[old-macro-stage-window]<https://user-images.githubusercontent.com/916140/80961355-b373a200-8e02-11ea-89aa-1b699784173f.png>

carandraug, 2020-06-29:
> Also maybe it should be forced positive as I assumed it was the current height and couldn’t work out what was changing it.

Hmm... Maybe the top position should not be allowed to be under the bottom position? I'm not sure what cockpit means with up/down, top/bottom, and touchdown on inverted and upright microscopes.