microscope-cockpit/cockpit #628: Wxpython 4.1.0 canvas size issue on Mac.

iandobbie, 2020-06-30:
With the crashing issues fixed we have a layout problem with wxpython 4.1.0 on the Mac. Teh OpenGL canvases all seem to only take up the bottom left hand quarter of the expected space. This happens with the camera viewpanel, the mosaic, macrostage and the touchscreen. I attach a screen shot of the camera viewpanel. Rescaling the image with the zoom functions works but vere fills more than the bottom left quarter. Resizing the window does not help as you get either blank space or push the image off the bottom of the window as it gets less than about 540 pixels. 
[image]

dstoychev, 2020-07-08:
In OpenGL there is the concept of a viewport, which defines the screen area used for drawing. For the XY microstage, we adjust the viewport on every OnPaint event:

https://github.com/MicronOxford/cockpit/blob/2cad2b9e23fc42c0925375995c4466efea186166/cockpit/gui/macroStage/macroStageXY.py#L164-L166

It seems unlikely, but maybe ``GetClientSize()`` function from wxWidgets returns improper values on Mac.

iandobbie, 2020-07-13:
Strangely enough a VM on the same machine with a fresh install of macos 10.15, python3.8.3 from python.org, then microscope and cockpit does not display this issue. Possible there is some other factors in play. Does anyone else have a mac and see what they see with wxpython4.1? Is it Catalina, python3.8, the vm screen interface?

carandraug, 2020-10-24:
I am checking now on a real Mac (not a VM), MacOSX 10.14.6, Python 3.8.6 (from python.org), and wxPython 4.1.0 (`4.1.0 osx-cocoa (phoenix) wxWidgets 3.1.4`),  and I don't see this problem. This is using a Dell monitor. Maybe the issue is with Apple displays (some "retina" hack?)

carandraug, 2020-10-24:
I tried with the large apple monitor that @iandobbie has on his desk (connected to the Apple mini on Mick's desk), and I also don't see this issue there.

iandobbie, 2020-10-25:
still having the same Just updated to python 3.8.6 with wxpython 4.1.0 on macos 10.14.6 still having the same issue. 

[image]

iandobbie, 2020-10-25:
Any suggestion of what I might to do fix this or other tests that could nail down the problems

iandobbie, 2020-10-26:
David has nailed it. It is something to do with the Apple "retina" displays on the laptops. I have discovered with an external monitor connected as the window moves between monitors the openGL canvas switches between the expected size on the external monitor and 1/4 the expected size on the laptops internal "retina" display. I suspect an underlying wx bug on mac as the OS scales things to 2x their normal size for the retina screen to enablle text to be readable on the 2,500 pixel wide laptop screen.

iandobbie, 2020-10-26:
I guess we need to create a minimal example in wx and post it to the wx bug list. I think it is this patch which causes this behaviour https://github.com/wxWidgets/wxWidgets/commit/f5d02e68088d9669ad95c32585dcbde24abd6465

iandobbie, 2021-01-25:
I have a fix that works on the mac. Currently only implemented on the imageviewer class, it takes account of the GetContentScaleFactor value to sacle the canvas width and height. It also needs to trap the EVT_DPI_CHANGED to do the right thing when moved between retina and normal screens. fix in https://github.com/iandobbie/cockpit/tree/mac-retina-fix

Can you please test to see it doesn't break other systems, eg windows and linux. If this works I will add similar logic to the other opengl canvases.

iandobbie, 2021-01-25:
The other openGL canvases do their scaling in the onpaint fiunction so dont need to trap the DPI change event. I have fixed the macrostage and  mosaic as well and it mostly works although the mosaic moves the visible region as you swap displays. Needs to rescale the zoom as we move between displays.

iandobbie, 2021-01-25:
I am pushing the rest of my changes, to my branch.

carandraug, 2021-01-25:
I just tested it on GTK and seems to work fine. Needs testing on windows now, ideally on a system that has a HDPI display.

dstoychev, 2021-01-26:
Tried it on a HDPI display under Windows. No errors and I did not notice any visual changes.

iandobbie, 2021-01-26:
Thanks Danny. I am going to fix the mosaic zoom issue which I think just requires trapping the change DPI event and using that to rescale the zoom value then pull it into master

iandobbie, 2021-01-26:
I faffed around for ages and could easily get a good remapping so it just keep the scale constant but recalculaltes displayed images etc. I think moving between low and high DPI displays is a rare event so a chaneg in the visible images in the mosaic due to the change in pixel number is acceptable.

closed 2021-01-26 by iandobbie