microscope-cockpit/cockpit #631: freetype

juliomateoslangerak, 2020-07-07:
In cockpit master freetype module is missing

```python
Traceback (most recent call last):
  File "C:/Users/julio/PycharmProjects/cockpit/cockpit/__init__.py", line 76, in <module>
    import cockpit.config
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\__init__.py", line 80, in <module>
    import cockpit.gui.mainWindow
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mainWindow.py", line 72, in <module>
    import cockpit.gui.fileViewerWindow
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\fileViewerWindow.py", line 53, in <module>
    from cockpit.gui import imageSequenceViewer
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageSequenceViewer.py", line 54, in <module>
    import cockpit.gui.imageViewer.viewCanvas
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 57, in <module>
    import cockpit.gui.freetype
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\freetype.py", line 57, in <module>
    import freetype
ModuleNotFoundError: No module named 'freetype'
```

juliomateoslangerak, 2020-07-07:
Sorry my mistake. I mixed with an error I got yesterday where it could not find gui/freetype and now it seems to be that freetype-py was not installed.

closed 2020-07-07 by juliomateoslangerak

carandraug, 2020-07-07:
For anyone coming here again, you will either need to install [freetype-py](https://pypi.org/project/freetype-py/) or rerun setup.py to get the new dependency installed.