microscope-cockpit/cockpit #632: Napari as a image viewer maybe as the mosaic window...

juliomateoslangerak, 2020-07-07:
I was wondering about the possibility to rely in napari as the infrastructure for visualizing images and maybe as the mosaic window... Would it be fast enough for the mosaic?
It would be great!
Definitely something to look into if moving from MRC file format to some zarr-ish file.

Can you label this as new feature

carandraug, 2020-07-07:
> I was wondering about the possibility to rely in napari as the infrastructure for visualizing images

Should be possible to replace the image viewer with napari. Do we want to do that?

> and maybe as the mosaic window... Would it be fast enough for the mosaic?

Not sure how that would work.

> It would be great!

Why?

> Definitely something to look into if moving from MRC file format to some zarr-ish file.

There's a dv plugin for napari so we wouldn't have to move from MRC to implement that (not that we shouldn't move away from MRC).

iandobbie, 2020-07-07:
Maybe @juliomateoslangerak should try add it as the view last file button and see how it works. Then we could think of extending it further.

juliomateoslangerak, 2020-07-08:
Yes, that was my idea. Fist as the last image viewer. We would get some extra and very neat visualization options