microscope-cockpit/cockpit #633: channel configurations are not validated

carandraug, 2020-07-07:
This come about on the discussion of issue #597 which is what validation should we do when loading channel files (the files loaded via the menu bar "Channels > Import channels ...":

> I'm wondering if we should be doing some validation of the loaded files.
>
> 1. if we load a file that tries to set devices that do not currently, should we error?
> 2. and the opposite, if we load a file that does not include devices that do exist, should we error?
>
> I think we should error in both cases above since the config file is obviously not adjusted to the current setup and we may end up setting everything incorrectly. @iandobbie and @juliomateoslangerak what do you think?

iandobbie, 2020-07-07:
I think if the config has a device which doesn't exist this should definitely throw and error. However I can see that adding a single device would then generate errors from everyones configs. How about just a warning in this situation.

iandobbie, 2020-07-07:
So I am suggesting
1) gives and error 
2) gives a warning

carandraug, 2020-08-29:
Done as @iandobbie suggested with bcc2825 . Closing as fixed.

closed 2020-08-29 by carandraug