microscope-cockpit/cockpit #638: sensible default saved top and bottom

carandraug, 2020-07-09:
The default saved top and bottom positions are 3010 and 3000 respectively, which only make sense in one of our microscopes (before 84471e6c1954538cddad they were 7780 and 7700 which only made sense on the original OMX at UCSF).

One possibility is to have the default saved top and bottom as the limits of the z stage but if these are not changed and someone accidentally presses go to bottom or go to top at the start, they might bash the objective against the stage. We could have both be the limit of the z stage furthest from the stage, but how do we do know which limit is that in both upright and inverted microscopes? I'm not clear what up/down, top/bottom, and touchdown does in cockpit for both cases.

carandraug, 2020-07-09:
The default values are currently defined in `cockpit.interfaces.stageMover`.

iandobbie, 2020-07-09:
Could we have a default of None which means effectively they are disabled, but with a config parameter so people can preset them on a system or user basis?