microscope-cockpit/cockpit #644: remove homebrew from macOS install instructions

carandraug, 2020-07-09:
@iandobbie The macOS instructions include installing homebrew to then install git, ftgl, and a newer python. If one is installing cockpit from PyPI, once it is there, they don't need git, macOS already comes with python installed, and cockpit no longer needs ftgl. Can we remove the whole homebrew stuff? Anyone that already uses homebrew won't need those instructions but if they don't use homebrew it seems a bit  too much to recommend installation of a package manager (no matter how useful we think it is for other things).

carandraug, 2020-07-09:
Reading online seems like python3 on macOS is installed as part of XCode so it's not that default. But there's also the installers in https://www.python.org/downloads/mac-osx/

iandobbie, 2020-07-10:
I believe that the default python on macos is still 2.7. With no need for ftgl we could go with the pythion.org installer but do we need to separately install freetype or does that just need the python stub code installed by pip?

iandobbie, 2020-07-10:
I can test this with the vm that I have. I'll try it later today and update the installer code.

carandraug, 2020-07-10:
> do we need to separately install freetype or does that just need the python stub code installed by pip?

It comes bundled on `freetype-py` which is listed as a dependency so it will be installed by pip.

iandobbie, 2020-07-13:
I have changed the macOS install instructions to use the python.org installer. One slight wrinkle is that the hidapi required by microscope need the xcode command line tools so I had to include a bit about that. If the xcode stuff isn't installed the pip2 install fails but displays a dialog prompting the user to install the relevant apple tools. After the install re-running the pip3 install succeeds and generates a working setup.

carandraug, 2020-07-13:
Sounds good to me. The macOS instructions were a subsection of the "Development sources" section so I moved them up a level so they are like the Linux and Windows instructions. Closing this.

closed 2020-07-13 by carandraug