microscope-cockpit/cockpit #645: Save image from camera view canvas fails

juliomateoslangerak, 2020-07-10:
Right click on the image canvas shows the save image. A dialog opens to save the image as dv but this fails:

```python
C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\guiUtils.py:304: ResourceWarning: unclosed file <_io.BufferedWriter name='C:\\Users\\julio\\Desktop\\test1\\0.50.dv'>
  frame.PopupMenu(menu, mousePosRelative)
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 429, in <lambda>
    self.Bind(wx.EVT_MENU, lambda event, action=action: action(), id=id)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\imageViewer\viewCanvas.py", line 834, in saveData
    cockpit.util.datadoc.writeDataAsMrc(self.imageData, path, wavelengths=wls)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\util\datadoc.py", line 697, in writeDataAsMrc
    handle.write(data_out)
ValueError: ndarray is not C-contiguous
```

carandraug, 2020-07-10:
Trying to replicate this issue it seems that it only happens when the camera is set with its transform to `(False, True, True)`, i.e., the image is flipped upside down and rotated 90 degrees. Is that right @juliomateoslangerak ?

iandobbie, 2020-07-10:
Wow that is some edge case!

carandraug, 2020-07-13:
I have just pushed 3d0d032a6b2d55740a2e28af76b84b37834f330b which fixes the issue for me and expands the testsuite. Closing as fixed.

closed 2020-07-13 by carandraug