microscope-cockpit/cockpit #647: SLM actions of 0 or 1 cause faulty action table

NickHallONBI, 2020-07-15:
Any experiment which adds an action which is a numerical 0 or 1 (such as line 246 of the structuredIllumination.py experiment) will cause a faulty line in the action table, since this action is never resolved. It is caught by the 

> elif action in [True, False]:

Statement at line 200 of the boulderSLM.py device and not converted to a trigger or otherwise resolved. This will cause the action table to have an action which corresponds neither to an analogue line voltage or a digital line True/False, therefore causing the action table to fail on the DSP.

carandraug, 2020-07-15:
> Any experiment which adds an action which is a numerical 0 or 1 (such as line 246 of the structuredIllumination.py experiment) will cause a faulty line in the action table, since this action is never resolved.

This is only true if the action is for the boulder SLM.

To add a bit more detail to this, `SIExperiment.generateActions` will add an action at the start to the SLM with a value of zero to trigger first new image:

```
            # Add a first trigger of the SLM to get first new image.
            table.addAction(curTime, self.slmHandler, 0)
```

which is then ignored by `BoulderSLM.examineActions` (`0 in [False]` returns `True` because the `in` operator tests for equality and `0 == False`):

```
        for i, (t, handler, action) in enumerate(table.actions):
[...]
            elif action in [True, False]:
                # Trigger action generated on earlier pass through.
                continue
```

That condition was added on 0655ae566ff36bfdf1f9a52348ca61f2ec53a8a6 but it's not clear to me when are the actions in an action table set to `True` or `False`. Anyone has any idea? We can fix the comparison of the action, but what if we just remove it if it's not needed?

iandobbie, 2020-07-17:
>         for i, (t, handler, action) in enumerate(table.actions):
> [...]
>             elif action in [True, False]:
>                 # Trigger action generated on earlier pass through.
>                 continue
> ```
> 
> That condition was added on [0655ae5](https://github.com/MicronOxford/cockpit/commit/0655ae566ff36bfdf1f9a52348ca61f2ec53a8a6) but it's not clear to me when are the actions in an action table set to `True` or `False`. Anyone has any idea? We can fix the comparison of the action, but what if we just remove it if it's not needed?

The action becomes True or False, when it is processed in the action table and becomes a digital signal. I seem to remember that actions can become reordered in the action table and so my guess is this prevents them being processed twice.

carandraug, 2020-07-17:
> The action becomes True or False, when it is processed in the action table and becomes a digital signal. I seem to remember that actions can become reordered in the action table and so my guess is [...]

I can't find where the action parameter is set to a bool and becomes the digital signal. Where is it?

carandraug, 2020-07-20:
After today's meeting, @NickHallONBI was going to find out if the action parameters are ever set to bool and if so, where.

NickHallONBI, 2020-11-04:
The action parameters are are set to bool by the table.addToggle function (line 282 in the boulderSLM.py script). This function can be found in the deviceHandler.py script, line 256-259:

`def addToggle(self, time, table):
    table.addAction(time, self, True)
    table.addAction(time + table.toggleTime, self, False)
    return time + table.toggleTime, 0`

As previously mentioned, any experiment which adds an action which is a numerical 0 or 1 (such as line 246 of the structuredIllumination.py experiment) is treated as a True/False by the `elif action in [True, False]:` check, resulting in a faulty entry. The solution we implement on DeepSIM is to change the elif check to:

`elif type(action) is bool:`

carandraug, 2020-11-04:
> The action parameters are are set to bool by the table.addToggle function (line 282 in the boulderSLM.py script).

But this happens *after* the check for the type of `action`. We roughly have:

```
class BoulderSLM:
    def examineActions(self, table):
        # ...
        for i, (t, handler, action) in enumerate(table.actions):
            # ...
            # this is the problematic check
            elif action in [True, False]:
            # ....
            # Many lines later, the lines that actually makes the check above relevant?
            for trig in range(numTriggers):
                t = table.addToggle(t, self.handler) # this adds bools to the table?
```

so how is that line introducing actions on the table that are being caught by the above conditional?

NickHallONBI, 2020-11-04:
It isn't adding actions that are caught by the conditional. If an action is caught by that conditional, that later line is skipped because the consequence of that conditional is `continue` which moves the for loop onto the next iteration (i.e. action). So the 'bad' lines with 0 or 1 as their actions are never replaced by that later line, get passed to the RedPitaya and cause a failure there.

carandraug, 2020-11-04:
I don't get it. That conditional is meant to catch cases where `action` is a bool (or that is the proposed changed here). If that condition will never be true, because the only code that creates bool actions is later, then what's the point of making that condition check for bools?

NickHallONBI, 2020-11-05:
The only entries this conditional catches are the "bad" entries. The addToggle function does append True False toggles to the end of the action table, but this occurs within the for loop. Since this for loop works over the actions enumerated at the start of loop, these latter appended actions are never seen. I think we can just get rid of this `elif` conditional all together.

carandraug, 2020-11-05:
Thank you for checking on hardware Nick. I've removed those lines now. Closing as fixed.

closed 2020-11-05 by carandraug