microscope-cockpit/cockpit #648: mosaic zoom in shold zoom on current mouse position not the center

carandraug, 2020-07-18:
When in the mosaic window, zooming in (by scrolling with the mouse) will zoom in the center. However, that's not how it usually works. When zooming, it should zoom in the current mouse position. TRy google maps for a more intuitive experience.

iandobbie, 2020-07-20:
Good point and the functionality to map window coords already exists so it should be easy to redo how this is done.

carandraug, 2024-01-02:
I've implemented this on my `648-zoom-at-point` branch but I'm not sure how it works with HDPI. @iandobbie can you test please? Zooming with mouse scroll keeps the mouse position fixed. Zooming with [Shif]t+[left mouse button down while moving mouse up/down] zooms in the centre.

iandobbie, 2024-01-02:
Just had a quick test and it isnt quite right. for just zooming (mac tracjpad with two fingers). On my laptop screen is seems the mouse position is off by a factor of 2, but on my main monitor it is fine. I'll have to go back and remember what the various factors are that control this scaling and how to apply the correction. 

If I do the shift, click and move the mouse pointer up and down it seems to work fine forboth screens, however I managed to get the following error.

```
Traceback (most recent call last):
  File “/Users/ID/src/cockpit/cockpit/gui/mosaic/canvas.py”, line 398, in onPaint
    self.overlayCallback()
  File “/Users/ID/src/cockpit/cockpit/gui/mosaic/window.py”, line 228, in drawOverlay
    self.scalebar = 100*(10**math.floor(math.log(1/self.canvas.scale,10)))
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: math domain error
```

presumably the canvas scale went negative and the log failed.

iandobbie, 2024-01-02:
I can't get the crash to happen again, I assume that it was a slight edge case, and maybe we should just test canvas.scale to check it is +ve and force it to be if it goes negative.

carandraug, 2024-01-02:
> Just had a quick test and it isnt quite right. for just zooming (mac tracjpad with two fingers). On my laptop screen is seems the mouse position is off by a factor of 2, but on my main monitor it is [...]

Does dividing `gl_xpoint` and `gl_ypoint` by `self.GetContentScaleFactor()` in ``MosaicCanvas.ZoomAtPoint` (line 427-428 at `cockpit/gui/mosaic/canvas.py`) work?  I would guess the fix for HDPI is somewhere there (in the conversion from wx to gl coordinates).

iandobbie, 2024-01-02:
Thanks for the hint. I will investigate. 

Ian

iandobbie, 2024-01-03:
Not that simple I'm afraid. I'm working on a proper fix..

iandobbie, 2024-01-03:
Ok, managed to fix the zoom around cursor with 
```
        width, height = self.GetClientSize() * self.GetContentScaleFactor()
        gl_xpoint = wx_point[0]*self.GetContentScaleFactor()
        gl_ypoint = ((wx_point[1]*self.GetContentScaleFactor() - height) * -1)
```
The two issues are it is *, not / and you need to not do this to the window as this is already scaled height, but now the zoom around center doesn't work as it is in effect scaled twice....
```
       width, height = self.GetClientSize() * self.GetContentScaleFactor()
       viewport_centre_point = (width /2, height /2)
       self.ZoomAtPoint(multiplier, viewport_centre_point)
```

iandobbie, 2024-01-03:
Ok, I think this is now working as expected. I checked it into my repository as branch 648-zoom-at-point

carandraug, 2024-01-03:
Thank you for testing. Ive squashed the changes and pushed to master. Closing as fixed.

closed 2024-01-03 by carandraug