microscope-cockpit/cockpit #649: Using Black as code formater

juliomateoslangerak, 2020-07-21:
What about using Black as a code formatter. 
It will reformat code and fix spacings, parenthesis,... while ensuring that the execution is not changed. 
I tried it and I like it. There are some things that I do not but I totally agree with the point of the creators: You get used to it and there is no further discussion. 

I love the visual homogeneity of the code across the project (and other projects if you use it else where)

juliomateoslangerak, 2020-07-21:
You may have a look in my master_blacked branch
https://github.com/juliomateoslangerak/cockpit.git

carandraug, 2020-07-25:
I'm fine with using Black in Cockpit (maybe we should do the same in Microscope). I would only change the default of 88 characters per line down to 79 though. Anyone has issues with passing the whole project though black and require future contributions to do the same?

carandraug, 2020-07-28:
We spoke about this yesterday on a meeting and agreed to do this after we merge some of the larger changes we have pending to reduce merge conflicts because of the code "blackening".

iandobbie, 2022-12-13:
This was discussed again at the development meeting and we will try to incorporate as much of Julio's and other outstanding code and then reformat with Black and add this as a single commit.

carandraug, 2022-12-14:
It was also discussed that we'd use the same settings as used in microscope, and we'd also use isort. For reference, the settings in microscope are at https://github.com/python-microscope/microscope/blob/master/pyproject.toml which are:

```
[tool.isort]
profile = "black"
line_length = 79
lines_after_imports = 2
# Despite the profile option being set to "black", there's still some
# options that are not correctly set in a compatible manner.  The
# following are for compatibility with black style.
combine_as_imports = true
include_trailing_comma = true
multi_line_output = 3  # multi lineoutput 3 is vert-hanging

[tool.black]
line-length = 79
target-version = ['py37', 'py38']
```