microscope-cockpit/cockpit #650: Actions plotter?

juliomateoslangerak, 2020-07-21:
I'm trying to debug things on my executor and I think it would be handy to have a plot of what you expect to see on the scope. 
Is there something like this?
I will make such a module if not. I was thinking to make a module in the util package. Does that seem to be a good place?

iandobbie, 2020-07-21:
This is a good idea, there is such a feature on the OMX software.

Whats your plan, interpret the lastActionTable and provide a plot with time against digital and a analogue signals?

juliomateoslangerak, 2020-07-22:
Yes, that's the idea. 
For the moment I added a plotting function to the ActionsTable class. 
https://github.com/juliomateoslangerak/cockpit/commit/74751b177076c6ca3d54b78933d8de341a801dc2
It should be beautified a bit but essentially works. I'm not sure how it should be implemented in the GUI. A new window that will call that method when opened?
[image]
[image]

juliomateoslangerak, 2020-12-07:
After the discussion last week. Do you think this can be merged into the Executors debugging window? Ian was suggesting a button 'Show last actions profile'. I'd love to do that but you are more into the GUI than me. Or it should jsut be the ocasion to get into it :o)

iandobbie, 2020-12-09:
This sounds like a great suggestion. Doesn't clutter the normal interface but allows access when you have an executor and might need it.

iandobbie, 2021-01-19:
I have started to integrate this into the dsp debug window in my branch https://github.com/iandobbie/cockpit/tree/action-plotter . 

A few outstanding issues
1) don't know whether to import the experiment at the top of the module or inline as it is at the moment
2) If you haven't yet run an experiment there is no action table so it generates an error.

iandobbie, 2021-01-19:
Updated to test for point 2 so it doesn't generate an error,

iandobbie, 2021-02-09:
This is a minimally invasive change. Adds a button to the dsp debug gui and the plotting functions to the actiontable file. Can I just pull this?

carandraug, 2021-02-11:
I think it's fine to import `cockpit.experiment.experiment` at the top of the module. I only have two issues with this:

1. if there's no last experiment to display, it prints a message to stdout. Instead, it should use a GUI message dialog (possibly `wx.MessageBox`).
2. The function to plot the action table does not require access to anything private on the actiontable. The act of plotting should be a separate function instead of a method (the same way that a numpy array does not have a method to display, instead different modules have functions or classes that will take an array and display it). Not having ` plotProfile` as a method of `ActionTable`  also means that one can then create an `ActionTable` without requiring matplotlib. I propose having `plotProfile` on the same place as the executor debug window.

I can do those changes if everyone else agrees with those changes.

iandobbie, 2021-02-11:
> I think it's fine to import `cockpit.experiment.experiment` at the top of the module. I only have two issues with this:
> 
> 1. if there's no last experiment to display, it prints a message to stdout. Instead, it should use a GUI message dialog (possibly `wx.MessageBox`).

Much better idea. My approach ewas very quick and dirty. 

> 2. The function to plot the action table does not require access to anything private on the actiontable. The act of plotting should be a separate function instead of a method (the same way that a [...]

So the plotProfile access experiment.lastExperiment.table and uses that data? Probably a better plan.

> I can do those changes if everyone else agrees with those changes.

Please feel free,. I am working on this digital Z stack idea. I have something that vaguely works in  cockpit. Now I need to implement an additional function in microscope to setup the digital Z stacks.

carandraug, 2021-02-11:
I have now pushed these changes. One thing I noticed is that this uses `matplotlib.pyplot` directly but in other places in Cockpit we create these specifying the correct backend. In my system, it seems to do the right. I wonder what's the reason to specify the backend and if we should do it here or if we should remove it from the others. Anyone?

Anyway, the thing is now merged so closing. Thank you

closed 2021-02-11 by carandraug