microscope-cockpit/cockpit #651: Fixing triggering timing on rolling shutter cameras

juliomateoslangerak, 2020-07-23:
Experiment must know if a camera has a global or rolling electronic shuttering mode in order to provide acurate timing and avoid the smearing effect of moving objects or inhomogeneous intensity.

I 'fixed' that in the past by adding a pseudoglobal exposure mode but:
- That seems inconsistent to me now as it is not an exposure mode and not something that has to be managed by microscope by rather by the an experiment that has to know the shuttering mode
- I did not fix the issue correctly (or it has meanwhile been broken) and in fast SI experiments, camera triggers were being missed cause the camera was not ready

So I have now changed this in two PRs in cockpit and microscope by:
- adding a micrsocopeCamera callback to request the shuttering mode
- I removed the pseudoglobal exposure mode (that was anyway only implemented in my branch of microscope
- Fixing the experiment that uses getShutteringMode to know when to trigger camera and light. I implemented the microscope method for Andors camera.

This is working for me and I believe it should not break anything. I do not know what happens with other cameras not overriding the getShuttering mode method.

juliomateoslangerak, 2020-07-23:
Sorry, I cannot manage to get a proper PR to work. These are the commits concerning cockpit
https://github.com/MicronOxford/cockpit/commit/3e3bd5b31ec64752e77e4e5486c321e22c8c1798
https://github.com/MicronOxford/cockpit/commit/aec15d81f7d29470b92b23e8d4ef6849270a9791
https://github.com/MicronOxford/cockpit/commit/8e14d2c4a110f31f1f9f9a273a1397ee9bfbb2c8

juliomateoslangerak, 2021-06-02:
Related to issue https://github.com/python-microscope/microscope/issues/209
I propose a fix for this in microscope and here. This fixes have been updated to the later changes in microscope.
The experiments are checking if a camera has global or rolling shutter and adapt the excitation timings accordingly. In the Zyla cameras the field is not homogeneous without this. 

Can you consider merging this. It is important.
This first commit works until the microscope version is updated on the cockpit side
https://github.com/juliomateoslangerak/cockpit/commit/daae6308c958038f4b679383986e596d1a2035e8
And with the hack removed
https://github.com/juliomateoslangerak/cockpit/commit/2d8dcb131d5090bcda8f6895684fd387109b480f

juliomateoslangerak, 2022-11-21:
There are a number of related issues that are lingering around and I'd like to centralize everything in here.

### microscope side
python-microscope/microscope#143 I initially suggest to implement a pseudoglobal triggering mode. This has been discarded. Not a good idea. There is a good deal of discussion there, but I suggest we close it.
python-microscope/microscope#209 Proposes a setter and getter for a electronic_shuttering property as well as implementation for the andor cameras. It is up to the user (cockpit) to deal with that information.

### cockpit side
Cockpit is supposed to deal with the information provided by the camera to adapt exposure timings to either global or rolling shutter conditions and prevent uneven detection artifacts.
#610 Opened by @carandraug. I never saw it so I opened a parallel issue (#651) I suggest we close it as duplicate
#651 Current issue. Reformulation of the same issue as here and proposed some commit that, for me, they solve the in-homogeneous illumination using short exposure timings. I suggest we recap the fixes here.
#665 is a similar issue but during the snap (take_image). A fix is proposed. Again I propose to close and recap here.

iandobbie, 2022-11-21:
Julio I think summarizing the issues in one place and then trying to close them is a good goal. Seems like you feel you have a complete fix although it may need support to work with other cameras. Is this correct?

juliomateoslangerak, 2022-11-21:
Yes, other cameras should implement this, at least for the cockpit part to work. Implementing the electronic shuttering mode in microscope should have no influence. 
We could wrap the calls in cockpit around a try-except NotImplementedError and choose a sensible default (global shutter currently).
I can have a look into the Ximea. 
Also, we might need some testing in real hardware.

There is a point I have probably underestimated along the way. I always use 'bulb' exposure mode, and that is the only scenario that I tested. Reading through the issues you mention that you use a trigger start pulse and you let the camera do the exposure time. A priory, that should not be a problem, but I guess I should look into things again taking this into account.

juliomateoslangerak, 2023-01-10:
I reassembled all the required changes into two fresh branches from master:
For changes on the microscope side get the commits from:
https://github.com/juliomateoslangerak/microscope/tree/Added_electronic_shuttering
For the corresponding changes in cockpit here:
https://github.com/juliomateoslangerak/cockpit/tree/Added_electronic_shuttering

I decorated _get_shuttering_mode as an abstractmethod and added the corresponding implementation for the Andor and Ximea cameras but it would have to be done for the other cameras too. If I'm not wrong these changes should not brake anything in cameras with a global shutter.

I tested these changes and they are working in my settup.

iandobbie, 2023-02-01:
This sounds like good work. Unfortunately I don't have a fully working setup with hardware triggers. I am working on setting it up, but probably in need of a better camera choice than the picam. We do have a flash 4 here that I might borrow from a system for some testing but I first need to make sure I have all the other stuff setup.