microscope-cockpit/cockpit #652: window menu show/hide

carandraug, 2020-07-23:
Currently the windows menu has the following entries for each window:

1. "Show/Hide" (toggles window between those two states)
2. "Raise to top" (brings a window to the top of all the other windows)
3. "Move to mouse" (moves a window to the mouse position, effectively the window menu)

There's a reason for each but I think we can simplify this and have only one item:

1. "Show/Hide" - it used to be that cockpit windows did not have a close box on the frame, and this menu item was the way to hide some of the windows. Now that all windows have a close box, there's no reason to have the hide functionality, it's easier and more intuitive for the user to just click on the window close box. This menu item can then simply become "Show" and the hide functionality can be removed.

2. "Raise to top" - This is useful for when a window is lost somewhere behind the other windows. So while technically not a hidden window, for the user this unhides the window. So this behaviour could be merged into "Show". if this is merged into "Show" the only difference is that calling "Show" on an already shown window will raise it to top too. However, a user calling "Show" likely also wants to interact with that window so raising it top is likely to be useful.

3. "Move to mouse" - This is mostly useful when the window is outside one of the monitors. This happens when adding or removing monitors (either at runtime or between sessions because cockpit will restore the window position but maybe that window position no longer exists). This functionality could also be merged on "Show", which would first check if the window position is outside the workspace and move it somewhere sensible.

With the proposed 3 changes, then we only need one menu item per window, and so could remove the submenus and only have a menu item with the window name.

For starters, I will turn "Show/hide" into "Show" and merge on it the "Raise to top". Would be nice to do the third but I'm unsure how other platforms behave about screen coordinates (someone wants to take on that?).

carandraug, 2020-07-23:
I have a branch that according to the documentation will move a window that is not on a valid display into the mouse position, which effectively merges the "Raise to top" menu item into the "Show" and then removes the whole submenu. Can someone on windows and OSX validate the changes? It's on my "wip-652-merge-move-to-mouse" branch.

juliomateoslangerak, 2020-07-24:
Did you merge this into master?
Raise to top is now gone in master. 
In windows, when I do move to mouse on a window that is not shown the window stays hidden. I think move to mouse should move to the mouse and then call show (show and bring to front)

carandraug, 2020-07-24:
> Did you merge this into master?

I merged in the removal of "Hide" from "Show/Hide" and the merging of "Raise to top" in "Show". I haven't merged in the changes of "Move to mouse" which is what needs testing in Mac and Windows (it's on my "wip-652-merge-move-to-mouse" branch).

> In windows, when I do move to mouse on a window that is not shown the window stays hidden. I think move to mouse should move to the mouse and then call show (show and bring to front)

The proposal here is to have show already do that if the window position is for some reason not valid (such as a display that has since been removed).

carandraug, 2020-08-28:
No one had any comments about this, so rebased and merged the proposed changes. Closing as fixed.

closed 2020-08-28 by carandraug