microscope-cockpit/cockpit #654: Lower status bar remains yellow

juliomateoslangerak, 2020-07-24:
When I run an experiment, the lower status bars get yellow and show the relevant info (nr of images and status), But when the experiment is done, although the text info is gone (it seems to understand that the experiment is done), the bars stay yellow.
I have this trace, though not sure it is related.
```python
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mainWindow.py", line 591, in _OnNewStatus
    assert len(event.EventData) == 2
AssertionError
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mainWindow.py", line 591, in _OnNewStatus
    assert len(event.EventData) == 2
AssertionError
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mainWindow.py", line 591, in _OnNewStatus
    assert len(event.EventData) == 2
AssertionError
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\mainWindow.py", line 591, in _OnNewStatus
    assert len(event.EventData) == 2
AssertionError
```

carandraug, 2020-07-24:
That trace shows that the `UPDATE_STATUS_LIGHT` was published with a number of arguments different from two. Basically, al publications of that event should be of this form: `events.publish(UPDATE_STATUS_LIGHT, light-name, message)`

Are you running the master branch? Or are you running the master branch with some of your changes on top? I have checked the current master and I don't see that event being called with incorrectly.

juliomateoslangerak, 2020-07-24:
I found the issue responsible for the trace and it does not seem to be the cause. The issue causing the trace is now fixed (there was a line where the event was published including the color to pass). 

So I get no trace but the status bar stays yellow.
I'm running my master branch. Which has the minimum differences with micron's master.

carandraug, 2020-07-24:
> I'm running my master branch. Which has the minimum differences with micron's master.

Can you replicate the issue on the master master branch? Or just on your master branch?

The logic that removes the yellow colour is on `StatusLights._OnNewStatus`. When all fields are empty, then the status bar colour goes back to grey. Maybe you have something that is setting the status light value to whitespace instead of an empty string?

juliomateoslangerak, 2020-07-27:
Yes. I can replicate it on micron's master. All dummy devices.

carandraug, 2020-07-27:
> Yes. I can replicate it on micron's master. All dummy devices.

It works on my side. Can you show your config and depot files, as well as the deviceserver config? What experiment and which settings?

juliomateoslangerak, 2020-07-27:
I made sure that no config or depot files are present in C:\Users\julio\AppData\Local\cockpit or C:\ProgramData\cockpit.
Cockpit creates all dummy cameras and lasers. 
I run a single z-stack experiment (see capture)
[image]
See that the status bar is still yellow from the last experiment

juliomateoslangerak, 2020-07-27:
I also get this error:
```python
Exception in thread Experiment-execute:
Traceback (most recent call last):
  File "C:\Users\julio\Applications\WPy64-3830\python-3.8.3.amd64\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\julio\Applications\WPy64-3830\python-3.8.3.amd64\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\experiment\experiment.py", line 382, in execute
    raise RuntimeError("Found a line that no executor could handle: %s" % str(self.table.actions[curIndex]))
RuntimeError: Found a line that no executor could handle: (Decimal('0'), <light source named Dummy 488 in group Dummy 488 light source>, True)
```

carandraug, 2020-07-27:
> See that the status bar is still yellow from the last experiment

The last experiment never ran because of the error you got (the dummy devices one gets on cockpit are not configured to run an experiment, but that's a separate issue).

Still, I can't replicate this issue with Python 3.7, wxPython 4.1.0, on GTK3. On `cockpit.gui.mainWindow.StatusLight._OnNewStatus` you have the following block:

```
        # This changes the colour of the whole bar, not only the
        # status (see issue #565).
        if any([self.GetStatusText(i) for i in range(self.FieldsCount)]):
            self.SetBackgroundColour(self._notificationColour)
        else:
            self.SetBackgroundColour(self._defaultBackgroundColour)
```

Can you add some print statements there and find what's happening? Maybe something like this:

```
        for i in range(self.FieldsCount)]):
            print("field %d has text '%s'" % (i, self.GetStatusText(i)))
        if any([self.GetStatusText(i) for i in range(self.FieldsCount)]):
            print('colour me notification')
            self.SetBackgroundColour(self._notificationColour)
        else:
            print('colour me default %s' % self._defaultBackgroundColour)
            self.SetBackgroundColour(self._defaultBackgroundColour)
```

juliomateoslangerak, 2020-07-28:
So I did. Here is the output wen running my master branch. If I get it well, the last two calls are as they are supposed to be and the problem must be downstream... that is in wx.
I'm running this in 3.6 (I went back cause my servers are in 3.6) but still in wx 4.1.0.

I will play around a bit to see if I can figure something more
```
Device configuration read from: ['C:\\Users\\julio\\AppData\\Local\\cockpit\\depot.conf']
Cockpit is running from C:\Users\julio\PycharmProjects\cockpit\cockpit
field 0 has text ''
colour me default (240, 240, 240, 255)
field 0 has text ''
field 1 has text ''
colour me default (240, 240, 240, 255)
field 0 has text ''
field 1 has text ''
colour me default (240, 240, 240, 255)
field 0 has text ''
field 1 has text ''
colour me default (240, 240, 240, 255)
field 0 has text ''
colour me default (240, 240, 240, 255)
field 0 has text ''
field 1 has text ''
colour me default (240, 240, 240, 255)
------- cockpit startup up to here -------

field 0 has text ''
field 1 has text 'Waiting for cRIO to finish'
colour me notification
field 0 has text ''
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 0/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text ''
field 1 has text 'Waiting for cRIO to finish'
colour me notification
field 0 has text ''
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 0/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 1/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 1/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 2/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 2/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 3/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 3/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 4/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 4/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 5/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 5/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 6/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 6/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 7/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 7/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 8/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 8/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 9/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 9/9'
field 1 has text 'Waiting for DSP to finish'
colour me notification
field 0 has text 'Red: 9/9'
field 1 has text 'Waiting for saving to complete'
colour me notification
field 0 has text 'Red: 9/9'
field 1 has text 'Waiting for saving to complete'
colour me notification
field 0 has text ''
field 1 has text 'Waiting for saving to complete'
colour me notification
field 0 has text ''
field 1 has text 'Waiting for saving to complete'
colour me notification
field 0 has text ''
field 1 has text ''
colour me default (240, 240, 240, 255)
field 0 has text ''
field 1 has text ''
colour me default (240, 240, 240, 255)
```

juliomateoslangerak, 2020-07-28:
If I replace the 'default' background color by an arbitrary color, that color is shown indeed at startup but it is not applied after the experiment run, even if the code is hit. That made me thing of a race condition in the event loop.
If I add a time.sleep just before the  ```self.SetBackgroundColour(self._defaultBackgroundColour)```
 everything works as expected and the status bar goes to default color.
But I really have to go to 3 sec to get it to work!!

juliomateoslangerak, 2020-07-28:
I can reproduce this in microns master (the experiment is running), but it is not very consistent. It does not work all the time.

carandraug, 2020-07-28:
I got a windows machine and managed to replicate the issue there. I have also found on the [wxWidgets documentation](https://docs.wxwidgets.org/3.1/classwx_window.html#a37219df52734626e23401fd83b25d8a0) what the issue was:

> The background colour is usually painted by the default wxEraseEvent event handler function under Windows and automatically under GTK. Note that setting the background colour does not cause an immediate refresh, so you may wish to call wxWindow::ClearBackground or wxWindow::Refresh after calling this function.

Basically, we just need to call `wx.Window.Refresh` at the end. I'm not sure why adding a sleep fixed the issue for you, maybe something else triggered a refresh. This explains why sometimes it works.

I pushed 0939122ccaec1e4eeebc95552fce5bb8e229515d to fix this. It also fixed the issue on Windows here so I'm closing this as fixed.

I also noticed that it appears that all the update light events are duplicated, but that is a separate issue.

closed 2020-07-28 by carandraug

juliomateoslangerak, 2020-07-28:
Sweet! it works. 
There is a typo in the commit 'Resfresh'

Thanks

carandraug, 2020-07-28:
> I also noticed that it appears that all the update light events are duplicated, but that is a separate issue.

This is not an issue. They are duplicated on the StatusLights class because there's two instances of it, one one on the main window and another on the touchscreen.

> There is a typo in the commit 'Resfresh'

Yeah,  I messed up. Was fixed on af09e54d7e4f1a6e65d8fd0829c37acde673613d