microscope-cockpit/cockpit #657: Freetype doesn't work on Windows 10

NickHallONBI, 2020-07-24:
freetype-py.2.2.0 for Python 3.7 is automatically installed when installing cockpit. However, the "import freetype" line in cockpit/gui/freetype.py fails with the following traceback:

> Traceback (most recent call last):
>   File "C:\Python37\lib\site-packages\freetype_py-2.2.0-py3.7-win-amd64.egg\freetype\raw.py", line 49, in <module>
>     _lib = ctypes.CDLL(filename)
>   File "C:\Python37\lib\ctypes\__init__.py", line 356, in __init__
>     self._handle = _dlopen(self._name, mode)
> OSError: [WinError 126] The specified module could not be found
> During handling of the above exception, another exception occurred:
> Traceback (most recent call last):
>   File "<input>", line 1, in <module>
>   File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.2.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 20, in do_import
>     module = self._system_import(name, *args, **kwargs)
>   File "C:\Python37\lib\site-packages\freetype_py-2.2.0-py3.7-win-amd64.egg\freetype\__init__.py", line 24, in <module>
>     from freetype.raw import *
>   File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.2.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 20, in do_import
>     module = self._system_import(name, *args, **kwargs)
>   File "C:\Python37\lib\site-packages\freetype_py-2.2.0-py3.7-win-amd64.egg\freetype\raw.py", line 52, in <module>
>     raise RuntimeError('Freetype library not found')
> RuntimeError: Freetype library not found

carandraug, 2020-07-24:
I would guess that this is an issue on freetype but I'm surprised because it's supposed to come with the dll. If you start a python shell (no pycharm), and just do `import freetype` does it work? If not, you should report this to https://github.com/rougier/freetype-py

NickHallONBI, 2020-07-24:
I get much the same error when importing it in a Python shell. I have raised an issue on the freetype GitHub

carandraug, 2020-07-27:
For reference, the issue opened with freetype-py is rougier/freetype-py/#126 . Closing this.

closed 2020-07-27 by carandraug