microscope-cockpit/cockpit #658: setlocale()  instead of wxLocale -problem (windows)

microphysics, 2020-07-25:
Cockpit seems to be unhappy with non-standard locale settings under  Windows:

Traceback (most recent call last):
  File "D:\PythonWin\Python38\lib\site-packages\cockpit\__init__.py", line 176, in OnInit
    module.makeWindow(main_window)
  File "D:\PythonWin\Python38\lib\site-packages\cockpit\gui\touchscreen.py", line 778, in makeWindow
    TSwindow = TouchScreenWindow(parent, title="Touch Screen view")
  File "D:\PythonWin\Python38\lib\site-packages\cockpit\gui\touchscreen.py", line 164, in __init__
    button = self.makeToggleButton(self.buttonPanel, 'Load/Unload',
  File "D:\PythonWin\Python38\lib\site-packages\cockpit\gui\touchscreen.py", line 430, in makeToggleButton
    bmp=self.checkBitmap(bitmap)
  File "D:\PythonWin\Python38\lib\site-packages\cockpit\gui\touchscreen.py", line 407, in checkBitmap
    bmp=wx.Bitmap(os.path.join( self.bitmapsPath, bitmap),
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1699) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

carandraug, 2020-07-25:
Cockpit does not really mess locales so I guess this is wxPython with Python 3.8 on windows issue.  You may be facing wxWidgets/Phoenix#1702 issue which will be fixed on wxPython 4.1.1. Can you install wxPython 4.1.1 from source and confirm this? Alternatively, can you test my "wip-658-port-phoenix-1702-PR" which backports the fix in wxPython into CockpitApp?

microphysics, 2020-07-27:
Neither wip-658-port-phoenix nor wxPython 4.1.1 fixed the issue.
Maybe it will just magically work on the computer in the lab, which I will try in the next couple of days.

Note: I installed wxPython 4.1.1 via  pip install -U --pre  -f https://wxpython.org/Phoenix/snapshot-builds/ wxPython instead of building it myself. I tried to build it, but apparently I am to dumb to install the correct part of Visual Studio.

carandraug, 2020-07-27:
This still looks like a wx issue. Can you run this minimal example on does it also fail because of locale?

```
import wx

path_to_some_png = '/tmp/foo.png'

app = wx.App()
bitmap = wx.Bitmap(path_to_some_png, wx.BITMAP_TYPE_ANY)
frame = wx.Frame(None)
frame.Show()
app.MainLoop()
```

Also, what do you get when you run this:

```
import locale
print('default is ', locale.getdefaultlocale())
print('locale is ', locale.getlocale())
```

microphysics, 2020-07-27:
py d:\test.py
Traceback (most recent call last):
  File "d:\test.py", line 6, in <module>
    bitmap = wx.Bitmap(path_to_some_png, wx.BITMAP_TYPE_ANY)
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1701) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

and

default is  ('en_DE', 'cp1252')
locale is  ('English_Germany', '1252')

carandraug, 2020-07-27:
Thanks. That's definitely a wxPython issue. Can you report with that minimal example to them, at https://github.com/wxWidgets/Phoenix/issues/new ?

microphysics, 2020-07-27:
done

microphysics, 2020-08-25:
Still fails:
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1701) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

carandraug, 2020-08-25:
On wxWidgets/Phoenix#1751 upstream suggests might be on the png that are being loaded for the icons. I'm not sure which one is causing the issue, can you add a print as they suggested to find which one is causing the error? And then we can investigate what's "funny" about the problematic png.

This is tricky to fix on our side since we can't reproduce the issue here.

microphysics, 2020-08-25:
It fails at the first PNG and changing the file doesn't seem to help.

I tried opening images in other formats with the minimal example in https://github.com/MicronOxford/cockpit/issues/658#issuecomment-664440244 and found that it "works" with BMP and JPG, although the image isn't shown, but fails with PNGs.

We don't have this issue on the computers that matter, I just thought it might be good to mention it so people don't run into this in the future, but if it is hard to reproduce and to debug, it'd be OK with postponing this until someone actually needs this fixed.

Borega, 2021-04-08:
has anyone found a sollution to this? It is rather crucial that I get it to run. I have exactly the situation the OP has.

carandraug, 2021-04-08:
It seems to only happen in Windows with a non english locale. We're unable to reproduce the issue as we don't have any such system (the versino of Windows we have only has english).

carandraug, 2021-04-08:
@Borega are you using Cockpit or are you asking because of the same "symptom" in another program?

@microphysics we have since completely rewritten the touchscreen and most of the icons are new. If the problem was on some png files, it is possible that the issue is now gone. Could you please check, on the computer that had this issue, if the issue is still present?

Borega, 2021-04-09:
Iam trying to use the wx Demo tool. But the problem is the same.

carandraug, 2021-04-09:
> Iam trying to use the wx Demo tool. But the problem is the same.

You're realy not using Cockpit then. Might be better to ask on wxPython's associated issue, i.e., wxWidgets/Phoenix#1751

Zakka-byte, 2021-09-05:
facing the same issue as above just after updating my python tools in Pycharm .Did you got the solution of above ??

AlexionCRM, 2022-06-23:
We had the same problem with Alexion CRM (build on python 3.9 an wx).

`############ Exception dialog shown to user #################
ExceptionDialog: Caught exception:
File "stdapp\main.py", line 191, in OnInit
File "stdapp\bootroutines\slipgatebootroutine.py", line 36, in execute
File "stdapp\bootroutines\slipgatebootroutine.py", line 56, in showSlipgateLogin
File "pynx\gui\slipgatelogin.py", line 52, in slipgateLogin
File "pynx\gui\slipgatelogindialog.py", line 51, in __init__
File "pynx\gui\icon.py", line 136, in getPngBitmap
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
##################################`

And were able to "solve" it by setting the correct language and regional settings in Windows (11). It had a mix of Languages "Nederlands (Nederland)" and "Nederlands (Belgie)" (I removed Belgie) and the regional settings were on "Aruba" which I set to Nederlands.

[image]

carandraug, 2026-11-10:
Upstream (wxPython) suggests this has been fixed since version 4.2.0 (released in August 2022) (see https://github.com/wxWidgets/Phoenix/issues/1751#issuecomment-6757703741 ). I remain without the means to check this as it is Windows specific. Closing.

closed 2026-11-10 by carandraug