microscope-cockpit/cockpit #659: Multisite experiment: checking sites called only first time

juliomateoslangerak, 2020-07-27:
When there are no sites marked or selected and I click on multi-site experiment I get the message that 'You must select points first'. Right. I cancel the experiment, I mark a few sites, select them and click again on multi-site experiment. I get the same message. Actually, getAllSites is not called a second time. I believe this might be cause the call is at instantiation of the EnumeratePannel and this panel must not be recreated...

juliomateoslangerak, 2020-07-27:
I just see that clicking the reset button is recreating the panel. 
hmmm this is not an obvious behaviour...

carandraug, 2020-07-27:
Since we never used the multi-site experiment, here's the steps to reproduce it:

1. without any sites marked on the mosaic window, click on the "Multi-site experiment" button
2. the "Sites to visit" text field on the "OMX multi-site experiment" window has the value "You must select points first"
3. close the "OMX multi-site experiment" dialog
4. mark some sites on the mosaic window
5. click the "Multi-site experiment" button again and note that "Sites to visit"  remains "You must select points first"
6. Click on the "reset" button which replaces "You must select points first" with the selected list of points