microscope-cockpit/cockpit #660: Customize light frequencies option overlays wrong place

juliomateoslangerak, 2020-07-27:
The customize light frequencies option in the multi-site experiment panel. overlays the new options on top of the window and over existing fields.

carandraug, 2020-07-27:
I can't replicate it under GTK3 (wxPython 4.1.0). What OS and wxPython are you running? Can you provide a screenshoot?

I have seen a similar issue on the single site experiment panel but intermittent, it seems to be dependent on the order one changes "Experiment type" but I haven't been abled to reproduce it reliably.

juliomateoslangerak, 2020-07-27:
I'm running on wx 4.1.0 (windows 10). Here you have the screenshots with and without the option
[image]
[image]

juliomateoslangerak, 2020-07-27:
Still on python 3.6... Will check if that matters.

carandraug, 2020-07-27:
Ah! I now see the "Customize light frequencies" button (I didn't before and thought you meant the "expose all cameras simultaneously"). I can reproduce it on GTK3, wxPython 4.1.0, and Python 3.7.

carandraug, 2020-07-27:
This is now fixed with 60b88b3f39fdedc4fbaaaf2049c5d9fe6c20da21 . It was a matter of calling `Layout` after showing/hiding the light frequencies panel.

closed 2020-07-27 by carandraug