microscope-cockpit/cockpit #664: Undefined handler for step-wise moving

juliomateoslangerak, 2020-07-28:
Pressing the keypad for moving the stage in steps does fail. The cause is an undefined handler.
```python
Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\keyboard.py", line 98, in onChar
    cockpit.interfaces.stageMover.step((0, 0, 1))
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\interfaces\stageMover.py", line 362, in step
    mover.Step(direction)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\interfaces\stageMover.py", line 267, in Step
    handler.moveRelative(step_size * sign)
NameError: name 'handler' is not defined
```

juliomateoslangerak, 2020-07-28:
Fixed in dab7e6895198e0d93dce99aecdb2ef8d804bd555

carandraug, 2020-07-28:
Thanks for the fix. Cherry-picked and pushed.

closed 2020-07-28 by carandraug