microscope-cockpit/cockpit #666: Unexpected SLM new menu entries

juliomateoslangerak, 2020-07-30:
Every time one sets the diffraction angle or generates a new sequence on the SLM a new menu item is added to the Windows menu.

juliomateoslangerak, 2020-07-31:
Yes, they are top-level windows. I don't know if that is what you mean.
This is what I see after setting twice diffraction angle and creating twice a sequence
[image]

carandraug, 2020-07-31:
That confirms it. The dialogs from `cockpit.gui.dialogs.getNumberDialog` are not destroyed after being used but they should. This can be tested without the boulder SLM, like so:

```
from cockpit.gui.dialogs import getNumberDialog
val = getNumberDialog.getNumberFromUser(None, 'Title', 'prompt', 77)
# Note that after been closed, there is a GetNumberDialog still around (it should not)
for w in wx.GetTopLevelWindows(): print(w)

# in comparison, if one uses the real wx GetX dialog, the dialog disappears after been used
import wx
val = wx.GetNumberFromUser('message', 'prompt', 'caption', 77)
for w in wx.GetTopLevelWindows(): print(w)
```

I guess one can fix the getNumberDialog but an alternative is to use one of the `wx.Get*FromUser`. There isn't one in wx to get a floating point value though.

carandraug, 2024-10-09:
Now fixed by calling `Destroy` after being done with the dialog. Closing as fixed.

closed 2024-10-09 by carandraug