microscope-cockpit/cockpit #667: Latest cockpit doesn't run with microscope.

iandobbie, 2020-08-17:
Just updated cockpit and microscope to master and thge test laser I have fails with

Traceback (most recent call last):
  File "/Users/ian/src/cockpit/cockpit/__init__.py", line 155, in OnInit
    main_window = cockpit.gui.mainWindow.makeWindow()
  File "/Users/ian/src/cockpit/cockpit/gui/mainWindow.py", line 662, in makeWindow
    window = MainWindow()
  File "/Users/ian/src/cockpit/cockpit/gui/mainWindow.py", line 474, in __init__
    panel = MainWindowPanel(self)
  File "/Users/ian/src/cockpit/cockpit/gui/mainWindow.py", line 214, in __init__
    lights_sizer.Add(mainPanels.LightControlsPanel(self), flag=wx.EXPAND)
  File "/Users/ian/src/cockpit/cockpit/gui/mainPanels.py", line 116, in __init__
    panel = LightPanel (self, light, power, filters)
  File "/Users/ian/src/cockpit/cockpit/gui/mainPanels.py", line 70, in __init__
    powCtrl.SetValue(lightPower.powerSetPoint *100.0)
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

And stderr ends with:

2020-08-17 11:47:19,797 ERROR    loggingWindow: 120    File "/Users/ian/src/cockpit/cockpit/devices/microscopeDevice.py", line 339, in _getPower

2020-08-17 11:47:19,797 ERROR    loggingWindow: 120      return self._proxy.power

2020-08-17 11:47:19,798 ERROR    loggingWindow: 120    File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/Pyro4/core.py", line 280, in __getattr__

2020-08-17 11:47:19,798 ERROR    loggingWindow: 120      raise AttributeError("remote object '%s' has no exposed attribute or method '%s'" % (self._pyroUri, name))

2020-08-17 11:47:19,799 ERROR    loggingWindow: 120  AttributeError: remote object 'PYRO:TestLaser@127.0.0.1:8002' has no exposed attribute or method 'power'

carandraug, 2020-08-17:
Did you also update microscope?

iandobbie, 2020-08-17:
Yup, just checked latest master as of 12:20 today, and reran setup.py as well just to confirm I am running the latest version.

carandraug, 2020-08-17:
I can't reproduce the issue. Can you give mode details on how to reproduce the issue? Also, did you restart the device server after updating microscope?

iandobbie, 2020-08-17:
I think the new python 3.8 install on jy mac is not running the develop code. I will investigate.

iandobbie, 2020-08-17:
A local install problem. Sorry for the issue.

closed 2020-08-17 by iandobbie