microscope-cockpit/cockpit #668: Binning error with andor cam on latets cockpit/microsocpe

iandobbie, 2020-08-17:
I am trying to upgrade the Oxford cryosim to the latest cockpit and microscope. Updated the remote microscope and the controlling machines cockpit. This generates the error below when I try to enable the camera. I suspect that the copy of microscope on the master machine is out of date compared to the remote version and  that produces the following error. 

Error in subscribed callable cockpit.devices.microscopeCamera.onObjectiveChange().  Traceback (most recent call last):
  File "c:\cockpit\cockpit\events.py", line 95, in publish
    func(*args, **kwargs)
  File "c:\cockpit\cockpit\devices\microscopeCamera.py", line 133, in onObjectiveChange
    self.updateTransform(transform)
  File "c:\cockpit\cockpit\devices\camera.py", line 84, in updateTransform
    self._setTransform((lr, ud, rot))
  File "c:\cockpit\cockpit\devices\microscopeCamera.py", line 112, in _setTransform
    self.updateSettings()
  File "c:\cockpit\cockpit\devices\microscopeCamera.py", line 106, in updateSettings
    self.settings.update(self._proxy.get_all_settings())
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 467, in _pyroInvoke
    data = serializer.deserializeData(msg.data, compressed=msg.flags & message.FLAGS_COMPRESSED)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\util.py", line 171, in deserializeData
    return self.loads(data)
  File "C:\Program Files\Python37\lib\site-packages\Pyro4\util.py", line 452, in loads
    return pickle.loads(data)
AttributeError: Can't get attribute 'Binning' on <module 'microscope' from 'C:\\Program Files\\Python37\\lib\\site-packages\\microscope-0.5.0-py3.7.egg\\microscope\\__init__.py'>

iandobbie, 2020-08-17:
Solved by upgrading the master side microscope install to match the remote version as cockpit assumes they are the same. Do we need versioning in microscope to prevent this?

carandraug, 2020-08-17:
> Do we need versioning in microscope to prevent this?

We have versioning in microscope. The problem is that his round of changes is not released yet (just like cockpit is not released). Once it is released, we can bump the version requirement.

closed 2020-08-17 by carandraug