microscope-cockpit/cockpit #669: takeImage assume there is an ambient source.

iandobbie, 2020-08-17:
The changes to the light sources to not have an ambient light source if other sources exist have broken the takeimage function.

I have the red pitaya executor, an andor ixon and 2 deepstar lasers, all real devices are remote from the cockpit computer using a single instance of microscope. Pressing snap on the GUI calls takeImage in the executor and generates the following error.

I think we should solve this in the first instance be re-instating the ambient light source,. It is also useful at times. 

 Traceback (most recent call last):
  File "c:\cockpit\cockpit\gui\mainWindow.py", line 124, in <lambda>
    snapButton.Bind(wx.EVT_BUTTON, lambda evt: cockpit.interfaces.imager.imager.takeImage())
  File "c:\cockpit\cockpit\interfaces\imager.py", line 156, in takeImage
    handler.takeImage()
  File "c:\cockpit\cockpit\handlers\imager.py", line 72, in takeImage
    self.callbacks['takeImage']()
  File "c:\cockpit\cockpit\handlers\executor.py", line 330, in takeImage
    if ambient.getIsEnabled():
AttributeError: 'NoneType' object has no attribute 'getIsEnabled'
Traceback (most recent call last):
  File "c:\cockpit\cockpit\gui\mainWindow.py", line 124, in <lambda>
    snapButton.Bind(wx.EVT_BUTTON, lambda evt: cockpit.interfaces.imager.imager.takeImage())
  File "c:\cockpit\cockpit\interfaces\imager.py", line 156, in takeImage
    handler.takeImage()
  File "c:\cockpit\cockpit\handlers\imager.py", line 72, in takeImage
    self.callbacks['takeImage']()
  File "c:\cockpit\cockpit\handlers\executor.py", line 330, in takeImage
    if ambient.getIsEnabled():
AttributeError: 'NoneType' object has no attribute 'getIsEnabled'

carandraug, 2020-08-17:
> The changes to the light sources to not have an ambient light source if other sources exist have broken the takeimage function. [...]
> I think we should solve this in the first instance be re-instating the ambient light source,. It is also useful at times.

The change was that the ambient light is not always there, you can still have it even if other light sources exist. You need to set it on the depot file.

iandobbie, 2020-08-17:
However the executor fails in snap or video mode if it doesn't exist.

iandobbie, 2020-08-17:
Adding a section to the depot file like

[ambient]
type: cockpit.device.light.AmbientLight

solves the snap issue but there is still a video issue I think

carandraug, 2020-08-18:
I can reproduce this issue now, the cameras need to be configured for triggering with one of the executors (instead of only soft triggers). Add something like this to `depot.conf`:

```
[south]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:TestCamera@127.0.0.1:8000
triggersource: dsp
triggerline: 3
```

carandraug, 2020-08-18:
Looking into this, the issue is this block which expects a light named "ambient":

```
        ambient = depot.getHandlerWithName('ambient')
        # If ambient light is enabled, extend exposure if necessary.
        if ambient.getIsEnabled():
            t = ambient.getExposureTime()
            if t > seq[-1][0]:
                seq.append((ambient.getExposureTime(), 0))
```

The executor prepares an action table for when to switch off the lights that it controls based on their exposure time. Then it looks for a light named "ambient" (which does not control), and extends the total time to wait before stopping camera exposure to include the ambient light.

The simplest fix is to check for the presence of a light handler named "ambient". However, can there be other light sources that are not controlled by the executor? If we see the reason for the special case behind "ambient" as not being controlled by the executor (it is not a digital client of the executor), can we generalise this and simply check for enabled light sources that are not digital clients of the executor?  Something like this:

```
        for light in depot.getHandlersOfType(depot.LIGHT_TOGGLE):
            if light not in self.digitalClients and light.getIsEnabled():
                # this light is enabled, extend exposure if necessary.
                t = light.getExposureTime()
                if t > seq[-1][0]:
                    seq.append((light.getExposureTime(), 0))
```

carandraug, 2020-08-25:
We spoke about this on Monday's meeting. @iandobbie said that the ambient light was a hack, but a very useful hack, and that he believes that just checking for lights not controlled by the executor should work just as well.

carandraug, 2020-08-25:
Looking into this, the handler created by an `AmbientLight` device is always named `"ambient"` and independent of the what the user actually named the device, and since we couldn't think of any other "light source" that shouldn't be controlled by the executor, then we can just get the handler with that name instead of iterating over all light sources each time we call `takeImage`.

So I pushed 587074e8694d which checks if the ambient handler exists. We can later change the logic to handle new cases as they come around. Closing as fixed.

closed 2020-08-25 by carandraug